rust-embedded/discovery #314: cargo doc => error: duplicate lang item in crate `panic_halt`: `panic_impl`.

ereichert, 2021-03-20:
I cloned the repo today and ran cargo doc on master in the /src/05-led-roulette directory.

`==> cargo doc
 Documenting aux5 v0.2.0 (/Users/ereichert/dev/tmp/discovery/src/05-led-roulette/auxiliary)
error: duplicate lang item in crate `panic_halt`: `panic_impl`.
  |
  = note: the lang item is first defined in crate `panic_itm` (which `aux5` depends on)
  = note: first definition in `panic_itm` loaded from /Users/ereichert/dev/tmp/discovery/target/thumbv7em-none-eabihf/debug/deps/libpanic_itm-eaa2b2ea5c1bfc9c.rmeta
  = note: second definition in `panic_halt` loaded from /Users/ereichert/dev/tmp/discovery/target/thumbv7em-none-eabihf/debug/deps/libpanic_halt-995f20f96388dea2.rmeta

error: aborting due to previous error

error: could not document `aux5`

Caused by:
  process didn't exit successfully: `rustdoc --edition=2018 --crate-type lib --crate-name aux5 src/05-led-roulette/auxiliary/src/lib.rs --target thumbv7em-none-eabihf -o /Users/ereichert/dev/tmp/discovery/target/thumbv7em-none-eabihf/doc --error-format=json --json=diagnostic-rendered-ansi -L dependency=/Users/ereichert/dev/tmp/discovery/target/thumbv7em-none-eabihf/debug/deps -L dependency=/Users/ereichert/dev/tmp/discovery/target/debug/deps --extern cortex_m=/Users/ereichert/dev/tmp/discovery/target/thumbv7em-none-eabihf/debug/deps/libcortex_m-af28499cfba8cbd0.rmeta --extern cortex_m_rt=/Users/ereichert/dev/tmp/discovery/target/thumbv7em-none-eabihf/debug/deps/libcortex_m_rt-36d5092093f3dd22.rmeta --extern panic_halt=/Users/ereichert/dev/tmp/discovery/target/thumbv7em-none-eabihf/debug/deps/libpanic_halt-995f20f96388dea2.rmeta --extern panic_itm=/Users/ereichert/dev/tmp/discovery/target/thumbv7em-none-eabihf/debug/deps/libpanic_itm-eaa2b2ea5c1bfc9c.rmeta --extern stm32f3_discovery=/Users/ereichert/dev/tmp/discovery/target/thumbv7em-none-eabihf/debug/deps/libstm32f3_discovery-ff375f2f78b055a4.rmeta --crate-version 0.2.0` (exit code: 1)`

Looks like there's a conflict with 2 upstream crates but wanted to see if anyone else knows the problem before I go down a rabbit hole.

I'm running using nightly on macos. 

rustc 1.50.0 (cb75ad5db 2021-02-10)

the-cat-that-codes, 2021-04-10:
Hi,

I had the same problem. As a workaround to allow the generation of the documentation, I commented the line with "panic-halt" in the Cargo.toml ...
I am new to Rust and my priority is (for now) to play with the discovery tutorial before entering into the details ...

eldruin, 2021-06-01:
This was fixed in #344

closed 2021-06-01 by eldruin