rust-embedded/discovery #319: Unable to compile the 05-led-roulette

hisho218, 2021-04-12:
I get this build error when I was going through chapter 5.1.

```
╰ cargo build --target thumbv7em-none-eabihf
   Compiling cast v0.2.4
error[E0432]: unresolved import `Error`
   --> /home/hishamo/.cargo/registry/src/github.com-1ecc6299db9ec823/cast-0.2.4/src/lib.rs:361:10
    |
361 |     use {Error, From};
    |          ^^^^^ no external crate `Error`

error[E0437]: type `Output` is not a member of trait `From`
   --> /home/hishamo/.cargo/registry/src/github.com-1ecc6299db9ec823/cast-0.2.4/src/lib.rs:192:21
    |
192 |                       type Output = $dst;
    |                       ^^^^^^^^^^^^^^^^^^^ not a member of trait `From`
...
364 | /     promotion! {
365 | |         i8    => f32, f64, i8, i16, i32, isize, i64;
366 | |         i16   => f32, f64,     i16, i32, isize, i64;
367 | |         i32   => f32, f64,          i32, isize, i64;
368 | |         isize => f32, f64,          i32, isize, i64;
369 | |         i64   => f32, f64,                      i64;
370 | |     }
    | |_____- in this macro invocation
    |
    = note: this error originates in a macro (in Nightly builds, run with -Z macro-backtrace for more info)
```

version info:
```
rustup check
stable-x86_64-unknown-linux-gnu - Up to date : 1.51.0 (2fd73fabe 2021-03-23)
nightly-x86_64-unknown-linux-gnu - Update available : 1.53.0-nightly (07e0e2ec2 2021-03-24) -> 1.53.0-nightly (a86612456 2021-04-11)
```

` rustc --version
rustc 1.51.0 (2fd73fabe 2021-03-23)
`

`╰ cargo --version
cargo 1.51.0 (43b129a20 2021-03-16)`

adamgreig, 2021-04-12:
Looks like this is caused by https://github.com/japaric/cast.rs/pull/27 which should hopefully be fixed soon. `cast` is coming from the stm32f3-discovery crate which depends on the lsm303dlhc crate which depends on cast; possibly you can temporarily patch this by putting `cast = "=0.2.2"` in the `src/05-led-roulette/Cargo.toml` file to try and force a dependency only on the older version.

hisho218, 2021-04-12:
I update the Cargo.toml file:

` cargo build --target thumbv7em-none-eabihf
   Compiling cast v0.2.3
error[E0463]: can't find crate for `std`
  |
  = note: the `thumbv7em-none-eabihf` target may not be installed

error: aborting due to previous error

For more information about this error, try `rustc --explain E0463`.
error: could not compile `cast`

To learn more, run the command again with --verbose.`

Now I am getting this failure:

`cargo build --target thumbv7em-none-eabihf
   Compiling cast v0.2.3
error[E0463]: can't find crate for `std`
  |
  = note: the `thumbv7em-none-eabihf` target may not be installed

error: aborting due to previous error

For more information about this error, try `rustc --explain E0463`.
error: could not compile `cast`

To learn more, run the command again with --verbose.`

This is strange because I did this

`rustup target add thumbv7em-none-eabihf`

yet the error state that 
` note: the `thumbv7em-none-eabihf` target may not be installed`

adamgreig, 2021-04-12:
Oh, my bad, you'll need to write:

```
cast = { version = "=0.2.3", default-features = false }
```

to disable the `std` feature, which is pulling in the `std` crate (which is not available on thumbv7).

hisho218, 2021-04-12:
That worked! Thanks Adam! :-)

hisho218, 2021-04-12:
Closing this issue since the suggested workaround resolve and the actual problem with the cast crate is being fixed here https://github.com/japaric/cast.rs/pull/27

closed 2021-04-12 by hisho218

sinclairnick, 2021-04-12:
Thanks! Was very close to giving up -- glad I found this