rust-embedded/discovery #320: Section 7.2. (mis)Optimization : Can not play the debug session as described in the tutorial

the-cat-that-codes, 2021-04-13:
Hi everyone,

I have some troubles at running this section correctly.

First, here are my version of rustup and rustc :

```console
$ rustup --version
rustup 1.23.1 (3df2264a9 2020-11-30)
info: This is the version for the rustup toolchain manager, not the rustc compiler.
info: The currently active `rustc` version is `rustc 1.53.0-nightly (07e0e2ec2 2021-03-24)`
```

The problem is that I can not execute step by step (using the gdb `next` instruction) the binary launched with `cargo run --release`. Here after is my session :

```console
$ cargo run --release
    Finished release [optimized + debuginfo] target(s) in 0.02s
     Running `arm-none-eabi-gdb -q -x openocd.gdb '/home/XXX/Dev/Rust/Embedded_Discovery/discovery-master/target/thumbv7em-none-eabihf/release/registers'`
Reading symbols from /home/XXX/Dev/Rust/Embedded_Discovery/discovery-master/target/thumbv7em-none-eabihf/release/registers...
0x08000252 in registers::__cortex_m_rt_main () at /rustc/07e0e2ec268c140e607e1ac7f49f145612d0f597/library/core/src/ptr/mod.rs:1150
1150	/rustc/07e0e2ec268c140e607e1ac7f49f145612d0f597/library/core/src/ptr/mod.rs: Aucun fichier ou dossier de ce type.
Loading section .vector_table, size 0x194 lma 0x8000000
Loading section .text, size 0x1220 lma 0x8000194
Loading section .rodata, size 0x1b4 lma 0x80013b4
Start address 0x8000194, load size 5480
Transfer rate: 13 KB/sec, 1826 bytes/write.
Breakpoint 1 at 0x8000224: file src/07-registers/src/main.rs, line 9.
Breakpoint 2 at 0x8001078
Breakpoint 3 at 0x80013b2
Note: automatically using hardware breakpoints for read-only addresses.

Breakpoint 1, main () at src/07-registers/src/main.rs:9
9	#[entry]
(gdb) s
halted: PC: 0x0800022a
registers::__cortex_m_rt_main () at src/07-registers/src/main.rs:11
11	    aux7::init();
(gdb) n
halted: PC: 0x08000254
halted: PC: 0x08000236
halted: PC: 0x0800023a
halted: PC: 0x0800023e
18	        ptr::write_volatile(GPIOE_BSRR as *mut u32, 1 << 9);
(gdb) n
halted: PC: 0x08000240
halted: PC: 0x08000244
halted: PC: 0x08000246
halted: PC: 0x0800024a
halted: PC: 0x0800024c
halted: PC: 0x08000250
halted: PC: 0x08000252
halted: PC: 0x08000252
halted: PC: 0x08000252
halted: PC: 0x08000252
halted: PC: 0x08000252
halted: PC: 0x08000252
halted: PC: 0x08000252
halted: PC: 0x08000252
halted: PC: 0x08000252
halted: PC: 0x08000252
halted: PC: 0x08000252
halted: PC: 0x08000252
halted: PC: 0x08000252
halted: PC: 0x08000252
halted: PC: 0x08000252
halted: PC: 0x08000252
halted: PC: 0x08000252
halted: PC: 0x08000252
halted: PC: 0x08000252
halted: PC: 0x08000252
halted: PC: 0x08000252
halted: PC: 0x08000252
halted: PC: 0x08000252
halted: PC: 0x08000252
halted: PC: 0x08000252
halted: PC: 0x08000252
^Chalted: PC: 0x08000252
halted: PC: 0x08000252

Program received signal SIGINT, Interrupt.
0x08000252 in registers::__cortex_m_rt_main () at /rustc/07e0e2ec268c140e607e1ac7f49f145612d0f597/library/core/src/ptr/mod.rs:1150
1150	/rustc/07e0e2ec268c140e607e1ac7f49f145612d0f597/library/core/src/ptr/mod.rs: Aucun fichier ou dossier de ce type.
```
after the second `next`instruction, I enter in the loop rather to go to the next `write_volatile` ...

If I disassemble, I get :
```console
Loading section .vector_table, size 0x194 lma 0x8000000
Loading section .text, size 0x1220 lma 0x8000194
Loading section .rodata, size 0x1b4 lma 0x80013b4
Start address 0x8000194, load size 5480
Transfer rate: 13 KB/sec, 1826 bytes/write.
Breakpoint 1 at 0x8000224: file src/07-registers/src/main.rs, line 9.
Breakpoint 2 at 0x8001078
Breakpoint 3 at 0x80013b2
Note: automatically using hardware breakpoints for read-only addresses.

Breakpoint 1, main () at src/07-registers/src/main.rs:9
9	#[entry]
(gdb) s
halted: PC: 0x0800022a
registers::__cortex_m_rt_main () at src/07-registers/src/main.rs:11
11	    aux7::init();
(gdb) n
halted: PC: 0x08000254
halted: PC: 0x08000236
halted: PC: 0x0800023a
halted: PC: 0x0800023e
18	        ptr::write_volatile(GPIOE_BSRR as *mut u32, 1 << 9);
(gdb) disassemble /m
Dump of assembler code for function registers::__cortex_m_rt_main:
10	fn main() -> ! {
   0x0800022a <+0>:	push	{r7, lr}
   0x0800022c <+2>:	mov	r7, sp

11	    aux7::init();
   0x0800022e <+4>:	bl	0x8000254 <aux7::init>
   0x08000232 <+8>:	movw	r0, #4120	; 0x1018
   0x08000236 <+12>:	mov.w	r1, #512	; 0x200
   0x0800023a <+16>:	movt	r0, #18432	; 0x4800
=> 0x0800023e <+20>:	str	r1, [r0, #0]
   0x08000240 <+22>:	mov.w	r1, #2048	; 0x800
   0x08000244 <+26>:	str	r1, [r0, #0]
   0x08000246 <+28>:	mov.w	r1, #33554432	; 0x2000000
   0x0800024a <+32>:	str	r1, [r0, #0]
   0x0800024c <+34>:	mov.w	r1, #134217728	; 0x8000000
   0x08000250 <+38>:	str	r1, [r0, #0]
   0x08000252 <+40>:	b.n	0x8000252 <registers::__cortex_m_rt_main+40>

End of assembler dump.
```
I have noticed that the file **mod.rs** is not found. Could it be the problem ? 
I have searched also in the directory that we can see in the tutorial (~/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/ptr/mod.rs) but it is not. May anyone explain me what I have missed ? 

Many thanks for your responses :-)

pcardune, 2021-04-30:
I've run into the same problem.

benschneider, 2021-05-23:
I see the individual string movements are there, try using stepi (or si) to step by machine instruction rather than line. Lights should then individually switch on and off.

NitinSaxenait, 2021-06-22:
Hi there @pcardune @benschneider @the-cat-that-codes I am also trying to debug in release mode I am using **cargo run --release** command in **discovery/src/07-registers** directory.
The error i am getting is-
`knoldus@knoldus-Vostro-3590:~/Documents/discovery/src/07-registers$ cargo run --release
    Finished release [optimized + debuginfo] target(s) in 0.04s
     Running `arm-none-eabi-gdb -q -x openocd.gdb /home/knoldus/Documents/discovery/target/thumbv7em-none-eabihf/release/registers`
error: could not execute process `arm-none-eabi-gdb -q -x openocd.gdb /home/knoldus/Documents/discovery/target/thumbv7em-none-eabihf/release/registers` (never executed)

Caused by:
  No such file or directory (os error 2)
`
I am using **rustup 1.24.2 (755e2b07e 2021-05-12)**

NitinSaxenait, 2021-06-23:
I am also working on the same chapter and facing the same issue in mis(optimization). Nothing comes this
`Breakpoint 1, main () at src/07-registers/src/main.rs:7
7       #[entry]
(gdb) s
registers::__cortex_m_rt_main () at src/07-registers/src/main.rs:9
9           aux7::init();
(gdb) n
16              ptr::write_volatile((GPIOE_BSRR as *mut u32), (1<<9));
`

dodsonmg, 2021-07-16:
I think this is an artefact of optimisation, which can create a bit of a discontinuity between your source and the assembly.  The debugger attempts to step through your source, but the source no longer sufficiently corresponds with the assembly and it gets a bit lost.  It still executes the program.  The repeated `halted: PC: 0x08000252` is the loop after it turns the LEDs off.

The simple solution, as @benschneider points out, is to just step through the assembly one instruction at a time with `stepi`.  You can view the assembly in the debugger as it's executed with `layout asm`.

@japaric: I'm happy to submit a PR, but there appear to be several stale issues and PRs associated with this section already (#238, #245, and #237 are the obvious ones).  Also, it looks like the most recent commit to discovery/src/07-registers/optimization.md ([cdf6be](https://github.com/rust-embedded/discovery/commit/cdf6be722d672e996893c52c793dae510b408566)) actually changed from stepping through the assembly to stepping through the source, so maybe we in this thread are simply doing something wrong...