rust-embedded/discovery #322: Unable to generate doc for aux5

hisho218, 2021-04-17:
Chapter 5.5. The Challange

When executing the command to generate the doc I get the following error:

```
cargo doc --open
 Documenting aux5 v0.2.0 (/home/hishamo/src/rust/discovery/src/05-led-roulette/auxiliary)
error: duplicate lang item in crate `panic_halt`: `panic_impl`.
  |
  = note: the lang item is first defined in crate `panic_itm` (which `aux5` depends on)
  = note: first definition in `panic_itm` loaded from /home/hishamo/src/rust/discovery/target/thumbv7em-none-eabihf/debug/deps/libpanic_itm-d20909402806804b.rmeta
  = note: second definition in `panic_halt` loaded from /home/hishamo/src/rust/discovery/target/thumbv7em-none-eabihf/debug/deps/libpanic_halt-ff0f80a7343a4c6b.rmeta

error: aborting due to previous error

error: could not document `aux5`

Caused by:
  process didn't exit successfully: `rustdoc --edition=2018 --crate-type lib --crate-name aux5 src/05-led-roulette/auxiliary/src/lib.rs --target thumbv7em-none-eabihf -o /home/hishamo/src/rust/discovery/target/thumbv7em-none-eabihf/doc --error-format=json --json=diagnostic-rendered-ansi -L dependency=/home/hishamo/src/rust/discovery/target/thumbv7em-none-eabihf/debug/deps -L dependency=/home/hishamo/src/rust/discovery/target/debug/deps --extern cortex_m=/home/hishamo/src/rust/discovery/target/thumbv7em-none-eabihf/debug/deps/libcortex_m-75ec4e1246590125.rmeta --extern cortex_m_rt=/home/hishamo/src/rust/discovery/target/thumbv7em-none-eabihf/debug/deps/libcortex_m_rt-0a659955e4a543e7.rmeta --extern panic_halt=/home/hishamo/src/rust/discovery/target/thumbv7em-none-eabihf/debug/deps/libpanic_halt-ff0f80a7343a4c6b.rmeta --extern panic_itm=/home/hishamo/src/rust/discovery/target/thumbv7em-none-eabihf/debug/deps/libpanic_itm-d20909402806804b.rmeta --extern stm32f3_discovery=/home/hishamo/src/rust/discovery/target/thumbv7em-none-eabihf/debug/deps/libstm32f3_discovery-72ec56457c541e0f.rmeta --crate-version 0.2.0` (exit code: 1)
```

dcabanis, 2021-04-18:
Same issue there

eldruin, 2021-04-19:
Maybe it would be interesting to add the doc generation to the CI

internet-diglett, 2021-04-29:
I'm also getting the same error. @hisho218 and @dcabanis, what OS are you running? I'm on an M1 Mac running Big Sur 11.3.

dcabanis, 2021-04-29:
I am on Ubuntu 20.04 here.

hisho218, 2021-04-29:
Distributor ID: Ubuntu
Description: Ubuntu 18.04.5 LTS
Release: 18.04
Codename: bionic

eldruin, 2021-06-01:
This was fixed in #344

closed 2021-06-01 by eldruin