rust-embedded/discovery #323: When to use modify() vs write()

wpd, 2021-04-19:
(I wish this project had a "Discussion" section, as this question really belongs there, not here as an "Issue".)
I am working through the examples in your book and am currently on chapter 8, "LEDs Again".  I appreciate how much of the work in that chapter you left as an "exercise to the reader", even if I might have grumbled a bit trying to figure out how to figure out what I needed to do.  But eventually, I got there, and I lit all 8 LEDs. :-)

But I noticed that my solution differed from yours in one significant detail... I called `gpioe.moder.write()` where you called `gpioe.moder.modify()`.  Reading the discussion in [The Embedded Rust Book](https://docs.rust-embedded.org/book/start/registers.html#modifying) about the `modify()`, function, I see "If we wish to change only one particular sub-field in this register and leave the other sub-fields unchanged, we can use the modify function."  That explains why you chose the `modify()` function.  But I would have naively assumed that a statement of the form:

`rcc.ahbenr.write(|w| w.iopeen().set_bit());`

would have implicitly read the AHBENR register and only manipulated the IOPEEN bit.  I notice that you called `write()` when you turned the LEDs on.

When/why should I use the `modify()` API when accessing a register field directly (such as via `iopeen()`, `moder8()`, or `odr11()`) vs using the `write()` API?  Or (to be blunt :-) ) why were you inconsistent in your use of the two API functions?

--wpd

eldruin, 2021-04-19:
Enabling discussions seems sensible to me for this project so I have enabled it now.
Although we have not discussed a r-e-wide policy about it.
@rust-embedded/resources any objections?

adamgreig, 2021-04-19:
Very happy to enable discussions.

wpd, 2021-04-19:
I have created the first Discussion topic with this question.  I am closing this as an issue now.

closed 2021-04-19 by wpd