rust-embedded/discovery #331: Can't compile any usart example

gravit22, 2021-04-29:
It panics right after this line
`core::mem::transmute(())`

[image]

gravit22, 2021-04-30:
Ok. I found information about pull request, and now it doesn't crash, but my minicom terminal doesn't show anything. 
[image]

sirhcel, 2021-05-31:
Hi @gravit22, several important changes have been landed in this repository in the last days. Could you please give it another try?

I'm using a STM32F3DISCOVERY MB1035D with its USB serial option and can successfully receive the _X_ characters from the [vanilla `11-uart` example](https://github.com/rust-embedded/discovery/blob/b6b90546a5e1b20041bf00417b7e8e081c420eec/src/11-usart/src/main.rs#L14). With the default configuration for the examples, PC4 gets used as TX and the communication parameters are 115,200 Baud, 8N1.

My sample output from minicom where I receive an _X_ for every program startup (press of the reset button):
```bash
$ minicom -o -D /dev/tty.usbmodem1211403 -b 115200

Welcome to minicom 2.8
[...]

XX
```

gravit22, 2021-06-04:
Hi.
When I try to run 
`minicom -o -D /dev/tty.usbmodem1211403 -b 115200`
It doesn't show anything, even minicom terminal

gravit22, 2021-06-04:
Hi. When I try `minicom -o -D /dev/tty.usbmodem1211403 -b 115200`
I minicom doesn't run.
I changed everything but it doesn't seams that it works. 
When I try minicom -D /dev/ttyACM0 -b 115200 the problem remains.

sirhcel, 2021-06-04:
> Hi. When I try `minicom -o -D /dev/tty.usbmodem1211403 -b 115200`
> minicom doesn't run.

Yes, this is my serial interface. And it looks that you've found `/dev/ttyACM0` to be yours and I assume that this is the device which gets reported by `dmesg` after you plug in your discovery board.

> I changed everything but it doesn't seams that it works.
> When I try minicom -D /dev/ttyACM0 -b 115200 the problem remains.

The example `11-uart` prints exactly one `X` upon each run of the firmware. After opening the serial device with minicom and pressing the reset button on the discovery board, you should receive exactly one `X`. Could you please build and run the actual [vanilla `11-uart` example](https://github.com/rust-embedded/discovery/blob/b6b90546a5e1b20041bf00417b7e8e081c420eec/src/11-usart/src/main.rs#L14) so that we can sort out other issues?

Do you have an oscilloscope at hand for checking the TX signal?

If you publish your current project on GitHub, I will have a look at it. Please publish a self-containing project like I've done with [`issue-246-interrupt-vectors`](https://github.com/sirhcel/discovery/tree/11d08a1edf3ae0366d273a0f93e552aa29e46f0e/src/issue-246-interrupt-vectors).

dodsonmg, 2021-07-19:
Probably an unnecessary suggestion, but just in case:

Have you removed the TX/RX jumper used for the loopback check in the previous section (10.3)?

al-jshen, 2021-08-24:
@dodsonmg Thank you! Couldn't get the `X` to show and was banging my head against the wall for the better part of an hour. Didn't even think to try this for some reason but it works!

rene0, 2021-10-31:
I also ended up removing the TX/RX wire between PC4 and PC5 after wondering why the X didn't show up, might be good to explicitly note this at the start of chapter 11 ?

closed 2022-02-06 by gravit22