rust-embedded/discovery #333: [11-usart] Cannot compile in release mode

Petrus97, 2021-05-01:
# The issue
Hi all, I tried to compile in release mode but I got this error:
```
warning: Linking globals named 'CORE_PERIPHERALS': symbol multiply defined!

error: failed to load bc of "cortex_m-b26f2bd747183185.cortex_m.87incrm8-cgu.0.rcgu.o": 

error: aborting due to previous error; 3 warnings emitted

error: could not compile `usart`
```
# What I found
Searching on the web the reason is that during the compilation there are multiple versions of `cortex-m` libraries. 
Running `cargo tree` here is what i found
```
[11-usart]$ cargo tree | grep cortex-m
│   ├── cortex-m v0.5.6
│   ├── cortex-m-rt v0.6.13
│   │   ├── cortex-m-rt-macros v0.1.8 (proc-macro)
│   │       ├── cortex-m v0.5.6 (*)
│   │       │   ├── cortex-m v0.5.6 (*)
│   │       │   ├── cortex-m-rt v0.6.13 (*)
│       └── cortex-m v0.7.2

```
As you can see I have two different version of `cortex-m` libs, this lib is used from `panic-itm` lib.
```
└── panic-itm v0.4.2
│       └── cortex-m v0.7.2
│           ├── bare-metal v0.2.5 (*)
│           ├── bitfield v0.13.2
│           ├── embedded-hal v0.2.5 (*)
│           └── volatile-register v0.2.0 (*)
```
Searching in the project folder in `Cargo.toml` of `auxiliary` lib I found these dependencies
```
[dependencies]
cortex-m = "=0.5.6" # 0.5.11 introduces a breaking change.  Use 0.5.6, since we know it works for this example
cortex-m-rt = "0.6.3"
panic-itm = "0.4.0"
```
The package version of `panic-itm` is `0.4.0` but during compilation I got `0.4.2`
```
   ...
   Compiling panic-itm v0.4.2
   ...
```
Looking at `panic-itm v0.4.2` source the `Cargo.toml` file we have 
```
[dependencies.cortex-m]
version = ">= 0.5.8, < 0.8"
```
In fact, [v0.4.0](https://crates.io/crates/panic-itm/0.4.0/dependencies) need the `v0.5.6 cortex-m`, but `cargo` doesn't downloaded it.
I tried manually but I got
```
[11-usart]$ cargo install panic-itm --version 0.4.0
    Updating crates.io index
error: specified package `panic-itm v0.4.0` has no binaries
```
### Just a question
Why when I compile in `debug-mode` it compiles fine? Does it compiles and put both `v0.5.6` and `v.0.7.2` cortex-m lib in the binary?

eisnstein, 2021-05-04:
I changed every `Cargo.toml` file, where the `panic-itm` dependency is defined to `panic-itm = "=0.4.0"`. That is in: 05, 06, 07, 08, 09, 11, 14, 15, 16. After that, I could compile 11-usart in release mode. I have not checked, if the others still work.

Petrus97, 2021-05-15:
Hi @eisnstein, thank you for the fix (sorry for the late answer). Now I can compile the 11-usart in release mode (I changed only the 05, 06 and 07) but changing the others `Cargo.toml` break the release mode compilation on them (In every chapter folder there is a different `cortex-m` version... 😐). For me it's not a big problem, now I'm curious about what happen during release mode compilation respect to debug mode. Before closing the issue can you indicate me some resources that explain this behaviour and why we need to modify the Cargo.toml in that way? Thank you in advance!

eisnstein, 2021-05-15:
No sorry I can't help you, I just tried to "fix" the panic-itm to version 0.4.0 with the "=", so that it does not update to a newer version. I am completely new to Rust and don't know how all this works.

sirhcel, 2021-05-31:
Hi @Petrus97, several changes and fixes have landed in this repository recently (for example https://github.com/rust-embedded/discovery/pull/341). They include switching to the actively maintained `stm32f3xx-hal`.

As of https://github.com/rust-embedded/discovery/commit/cc254bbbb381931e22e5f491ad48e558cc5f940d I can build all examples successfully even in release mode.
```bash
$ git describe --always --dirty
cc254b
$ rustc --version
rustc 1.52.1 (9bc8c42bb 2021-05-09)
$ cargo build --release --target thumbv7em-none-eabihf 
```
Yes, there are still warnings about unused variables, constants, ... But the build itself works.

I expect this to resolve your issues too. Does the latest updates work out for you too?

closed 2022-10-10 by Petrus97