rust-embedded/discovery #336: I can't run i2c examples

gravit22, 2021-05-07:
I got this:
`error: could not execute process `arm-none-eabi-gdb -q -x openocd.gdb target/thumbv7em-none-eabihf/debug/i2c` (never executed)

Caused by:
  No such file or directory (os error 2)
`

I can't run everything above that tutorial, but I can run the tutorials like led_rullete.

eldruin, 2021-05-29:
You may have been in the wrong folder. Have you checked that the file exists in the path you provide to `arm-none-eabi-gdb`?

sirhcel, 2021-06-02:
Hello @gravit22, do you have a GDB for ARM installed and in your search path? What does `arm-none-eabi-gdb --version` give you?

I have `arm-none-eabi-gdb` 10.1.90.20201028-git in my search path and I am able to build and start this example (though I have not played around with it any further). I'm starting a GDB server in a separate terminal:
```
$ openocd -f interface/stlink-v2-1.cfg -f target/stm32f3x.cfg
[...]
Info : STLINK [...] (API v2) VID:PID 0483:374B
Info : Target voltage: 2.892163
Info : stm32f3x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : Listening on port 3333 for gdb connections
```
And after a successful build I can start the example which drops me into a GDB session:
```
discovery/src/14-i2c $ cargo run --target thumbv7em-none-eabihf
[...]
    Finished dev [unoptimized + debuginfo] target(s) in 0.05s
     Running `arm-none-eabi-gdb -q -x openocd.gdb [...]/discovery/target/thumbv7em-none-eabihf/debug/i2c`
Reading symbols from [...]/discovery/target/thumbv7em-none-eabihf/debug/i2c...
i2c::__cortex_m_rt_main () at src/14-i2c/src/main.rs:44
44          loop {}
[...]
Breakpoint 3, i2c::__cortex_m_rt_main_trampoline () at src/14-i2c/src/main.rs:15
15      #[entry]
(gdb)
```

gravit22, 2021-06-04:
Hello. Whne I run `gdb-multiarch --version` I get:
`GNU gdb (Ubuntu 10.1-2ubuntu2) 10.1.90.20210411-git
Copyright (C) 2021 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
`
I don't have arm-none-eabi-gdb

sirhcel, 2021-06-04:
Hello @gravit22, your adaption to `gdb-multiarch` looks good at a first glance. Could you please give the complete output of your attempt of running the firmware?

Could you please publish your code including your adaption for `gdb-multiarch` here on GitHub and link it from this issue? The formatting in https://github.com/rust-embedded/discovery/issues/336#issuecomment-854623201 looks a bit strange to me as I would not have expected to read a line with `runner = "gdb -q"`.