rust-embedded/discovery #338: Chapter 11 doesn't compile

insomnimus, 2021-05-21:
Hi, first I would  like to thank the author for the book, it's a nice intro to embedded I needed.

I've followed every chapter and until chapter 11, all worked just fine.

Some info:

-  OS: Windows 10
-  Cargo version: cargo 1.52.0 (69767412a 2021-04-21)
-  rustc version: rustc 1.52.1 (9bc8c42bb 2021-05-09)

Building the code in src/11-usart errors with this:
`error: linking with `rust-lld` failed: exit code: 1`

here's the full output

```sh
PS src/11-usart> cargo build --verbose --target thumbv7em-none-eabihf
       Fresh semver-parser v0.7.0
       Fresh ucd-trie v0.1.3
       Fresh unicode-xid v0.2.2
       Fresh version_check v0.9.3
       Fresh vcell v0.1.3
       Fresh nb v1.0.0
       Fresh void v1.0.2
       Fresh stable_deref_trait v1.2.0
       Fresh aligned v0.2.0
       Fresh r0 v0.2.2
       Fresh bitfield v0.13.2
       Fresh byteorder v1.4.3
       Fresh semver v0.9.0
       Fresh pest v2.1.3
       Fresh volatile-register v0.2.0
       Fresh nb v0.1.3
       Fresh typenum v1.13.0
       Fresh rustc_version v0.2.3
       Fresh semver-parser v0.10.2
       Fresh proc-macro2 v1.0.27
       Fresh hash32 v0.1.1
       Fresh embedded-hal v0.2.5
       Fresh generic-array v0.11.2
       Fresh generic-array v0.12.4
       Fresh generic-array v0.13.3
       Fresh semver v0.11.0
       Fresh quote v1.0.9
       Fresh generic-array v0.14.4
       Fresh l3gd20 v0.2.0
       Fresh heapless v0.3.7
       Fresh rustc_version v0.3.3
       Fresh syn v1.0.72
       Fresh as-slice v0.1.5
       Fresh bare-metal v0.2.5
       Fresh cortex-m-rt-macros v0.1.8
       Fresh aligned v0.3.4
       Fresh cortex-m-rt v0.6.14
       Fresh cortex-m v0.5.6
       Fresh cortex-m v0.6.4
       Fresh cast v0.2.6
       Fresh stm32f30x v0.7.1
       Fresh panic-itm v0.4.2
       Fresh stm32f30x-hal v0.2.0
       Fresh lsm303dlhc v0.2.0
       Fresh f3 v0.6.1
       Fresh aux11 v0.1.0 (D:\home\books\discovery\src\11-usart\auxiliary)
   Compiling usart v0.1.0 (D:\home\books\discovery\src\11-usart)
     Running `rustc --crate-name usart --edition=2018 src\11-usart\src\main.rs --error-format=json --json=diagnostic-rendered-ansi --crate-type bin --emit=dep-info,link -C embed-bitcode=no -C debuginfo=2 -C metadata=e74669927cc7cdae -C extra-filename=-e74669927cc7cdae --out-dir D:\home\books\discovery\target\thumbv7em-none-eabihf\debug\deps --target thumbv7em-none-eabihf -C incremental=D:\home\books\discovery\target\thumbv7em-none-eabihf\debug\incremental -L dependency=D:\home\books\discovery\target\thumbv7em-none-eabihf\debug\deps -L dependency=D:\home\books\discovery\target\debug\deps --extern aux11=D:\home\books\discovery\target\thumbv7em-none-eabihf\debug\deps\libaux11-712ea8d94dbc61cf.rlib --extern heapless=D:\home\books\discovery\target\thumbv7em-none-eabihf\debug\deps\libheapless-44e3f879ab01396c.rlib -C link-arg=-Tlink.x -L D:\home\books\discovery\target\thumbv7em-none-eabihf\debug\build\cortex-m-cb27bef307bc3107\out -L D:\home\books\discovery\target\thumbv7em-none-eabihf\debug\build\cortex-m-rt-dc7a6cfda80ec4d9\out -L D:\home\books\discovery\target\thumbv7em-none-eabihf\debug\build\f3-f71ec4a17674443f\out -L D:\home\books\discovery\target\thumbv7em-none-eabihf\debug\build\stm32f30x-a1db1d15921ecbc5\out -L D:\home\books\discovery\target\thumbv7em-none-eabihf\debug\build\cortex-m-1004092a06c0760a\out`
warning: unused variable: `mono_timer`
  --> src\11-usart\src\main.rs:10:18
   |
10 |     let (usart1, mono_timer, itm) = aux11::init();
   |                  ^^^^^^^^^^ help: if this is intentional, prefix it with an underscore: `_mono_timer`
[... 1 lines cut ...]
   = note: `#[warn(unused_variables)]` on by default

warning: unused variable: `itm`
  --> src\11-usart\src\main.rs:10:30
   |
10 |     let (usart1, mono_timer, itm) = aux11::init();
   |                              ^^^ help: if this is intentional, prefix it with an underscore: `_itm`

error: linking with `rust-lld` failed: exit code: 1
  |
  = note: "rust-lld" "-flavor" "gnu" "--eh-frame-hdr" "-L" "C:\\Users\\insomnia\\.rustup\\toolchains\\stable-x86_64-pc-windows-msvc\\lib\\rustlib\\thumbv7em-none-eabihf\\lib" "D:\\home\\books\\discovery\\target\\thumbv7em-none-eabihf\\debug\\deps\\usart-e74669927cc7cdae.usart.9359hgya-cgu.0.rcgu.o" "D:\\home\\books\\discovery\\target\\thumbv7em-none-eabihf\\debug\\deps\\usart-e74669927cc7cdae.usart.9359hgya-cgu.1.rcgu.o" "D:\\home\\books\\discovery\\target\\thumbv7em-none-eabihf\\debug\\deps\\usart-e74669927cc7cdae.usart.9359hgya-cgu.2.rcgu.o" "D:\\home\\books\\discovery\\target\\thumbv7em-none-eabihf\\debug\\deps\\usart-e74669927cc7cdae.usart.9359hgya-cgu.3.rcgu.o" "-o" "D:\\home\\books\\discovery\\target\\thumbv7em-none-eabihf\\debug\\deps\\usart-e74669927cc7cdae" "--gc-sections" "-L" "D:\\home\\books\\discovery\\target\\thumbv7em-none-eabihf\\debug\\deps" "-L" "D:\\home\\books\\discovery\\target\\debug\\deps" "-L" "D:\\home\\books\\discovery\\target\\thumbv7em-none-eabihf\\debug\\build\\cortex-m-cb27bef307bc3107\\out" "-L" "D:\\home\\books\\discovery\\target\\thumbv7em-none-eabihf\\debug\\build\\cortex-m-rt-dc7a6cfda80ec4d9\\out" "-L" "D:\\home\\books\\discovery\\target\\thumbv7em-none-eabihf\\debug\\build\\f3-f71ec4a17674443f\\out" "-L" "D:\\home\\books\\discovery\\target\\thumbv7em-none-eabihf\\debug\\build\\stm32f30x-a1db1d15921ecbc5\\out" "-L" "D:\\home\\books\\discovery\\target\\thumbv7em-none-eabihf\\debug\\build\\cortex-m-1004092a06c0760a\\out" "-L" "C:\\Users\\insomnia\\.rustup\\toolchains\\stable-x86_64-pc-windows-msvc\\lib\\rustlib\\thumbv7em-none-eabihf\\lib" "-Bstatic" "D:\\home\\docs\\books\\discovery\\target\\thumbv7em-none-eabihf\\debug\\deps\\libaux11-712ea8d94dbc61cf.rlib" "D:\\home\\docs\\books\\discovery\\target\\thumbv7em-none-eabihf\\debug\\deps\\libf3-0325f235f8066bca.rlib" "D:\\home\\docs\\books\\discovery\\target\\thumbv7em-none-eabihf\\debug\\deps\\libstm32f30x_hal-b449d3f499dfc1c8.rlib" "D:\\home\\docs\\books\\discovery\\target\\thumbv7em-none-eabihf\\debug\\deps\\libstm32f30x-6a1f7a9a3c52dbe7.rlib" "D:\\home\\docs\\books\\discovery\\target\\thumbv7em-none-eabihf\\debug\\deps\\liblsm303dlhc-3f9f0349d6342d52.rlib" "D:\\home\\docs\\books\\discovery\\target\\thumbv7em-none-eabihf\\debug\\deps\\libcast-f677559607a68661.rlib" "D:\\home\\docs\\books\\discovery\\target\\thumbv7em-none-eabihf\\debug\\deps\\libl3gd20-dc80c510e5afe293.rlib" "D:\\home\\docs\\books\\discovery\\target\\thumbv7em-none-eabihf\\debug\\deps\\libgeneric_array-dbfcfa3c8302c872.rlib" "D:\\home\\docs\\books\\discovery\\target\\thumbv7em-none-eabihf\\debug\\deps\\libembedded_hal-3f8414a19ee6fb5c.rlib" "D:\\home\\docs\\books\\discovery\\target\\thumbv7em-none-eabihf\\debug\\deps\\libvoid-b20ebbf8be9b2a6a.rlib" "D:\\home\\docs\\books\\discovery\\target\\thumbv7em-none-eabihf\\debug\\deps\\libnb-85cedffd0abd4992.rlib" "D:\\home\\docs\\books\\discovery\\target\\thumbv7em-none-eabihf\\debug\\deps\\libnb-a84b77a04f6df8dc.rlib" "D:\\home\\docs\\books\\discovery\\target\\thumbv7em-none-eabihf\\debug\\deps\\libcortex_m_rt-6ea0b6702bb57ae7.rlib" "D:\\home\\docs\\books\\discovery\\target\\thumbv7em-none-eabihf\\debug\\deps\\libr0-0836fe5a1e343e87.rlib" "D:\\home\\docs\\books\\discovery\\target\\thumbv7em-none-eabihf\\debug\\deps\\libcortex_m-a7292bcbde62a084.rlib" "D:\\home\\docs\\books\\discovery\\target\\thumbv7em-none-eabihf\\debug\\deps\\libaligned-2080b53a751b9e30.rlib" "--start-group" "D:\\home\\docs\\books\\discovery\\target\\thumbv7em-none-eabihf\\debug\\deps\\libpanic_itm-687622c341dcffd5.rlib" "D:\\home\\docs\\books\\discovery\\target\\thumbv7em-none-eabihf\\debug\\deps\\libcortex_m-2b2954fd14f634d9.rlib" "D:\\home\\docs\\books\\discovery\\target\\thumbv7em-none-eabihf\\debug\\deps\\libvolatile_register-672016c479e83098.rlib" "D:\\home\\docs\\books\\discovery\\target\\thumbv7em-none-eabihf\\debug\\deps\\libvcell-60f2575b8d46b981.rlib" "D:\\home\\docs\\books\\discovery\\target\\thumbv7em-none-eabihf\\debug\\deps\\libbare_metal-ab9b1499b4d5c071.rlib" "D:\\home\\docs\\books\\discovery\\target\\thumbv7em-none-eabihf\\debug\\deps\\libaligned-5d804b92519dc883.rlib" "D:\\home\\docs\\books\\discovery\\target\\thumbv7em-none-eabihf\\debug\\deps\\libas_slice-0581e1b2377ca1a9.rlib" "D:\\home\\docs\\books\\discovery\\target\\thumbv7em-none-eabihf\\debug\\deps\\libstable_deref_trait-cce8b255d01405d8.rlib" "D:\\home\\docs\\books\\discovery\\target\\thumbv7em-none-eabihf\\debug\\deps\\libgeneric_array-21f01d8096092e56.rlib" "D:\\home\\docs\\books\\discovery\\target\\thumbv7em-none-eab [... cut ...] ome\books\discovery\target\thumbv7em-none-eabihf\debug\build\cortex-m-cb27bef307bc3107\out -L D:\home\books\discovery\target\thumbv7em-none-eabihf\debug\build\cortex-m-rt-dc7a6cfda80ec4d9\out -L D:\home\books\discovery\target\thumbv7em-none-eabihf\debug\build\f3-f71ec4a17674443f\out -L D:\home\books\discovery\target\thumbv7em-none-eabihf\debug\build\stm32f30x-a1db1d15921ecbc5\out -L D:\home\books\discovery\target\thumbv7em-none-eabihf\debug\build\cortex-m-1004092a06c0760a\out` (exit code: 1)
```

insomnimus, 2021-05-21:
Note: changing cortex-m version to "=0.6.4" in auxiliary/Cargo.toml didn't fix the issue as well.

dzvon, 2021-05-26:
Same problem.

commit c44e72c1 (2021-05-29): Merge #341

341: Fix compile error r=eldruin a=dzvon

Fix #338 #340 

It looks like these problems are caused by multiple versions. I have no idea of the underlying reasons, but I think the simplest workaround is to keep one version. So I remove f3 crate which depends on `cortex-m v0.5.6` to fix it.

There are also the same problems in chapter 11,14,15. But it seems more complex, as I'm a newbie in this field, and I'm still learning :-). So I only submit this to fix chapter 9. Thanks a lot.
PR #341 (2021-05-29): Fix compile error
Fix #338 #340 

It looks like these problems are caused by multiple versions. I have no idea of the underlying reasons, but I think the simplest workaround is to keep one version. So I remove f3 crate which depends on `cortex-m v0.5.6` to fix it.

There are also the same problems in chapter 11,14,15. But it seems more complex, as I'm a newbie in this field, and I'm still learning :-). So I only submit this to fix chapter 9. Thanks a lot.
closed 2021-05-29 by bors[bot]