rust-embedded/discovery #374: I am working in embedded development using rust and tried to use itm but its not getting installed

NitinSaxenait, 2021-07-15:
Hi i was working with discovery book and I was trying to install the itmdump but it's showing me error.
error: there is nothing to install in itm v0.3.1, because it has no binaries cargo installis only for installing programs, and can't be used with libraries. To use a library crate, add it as a dependency in a Cargo project instead.
rustc Version
ustc 1.55.0-nightly (2f391da2e 2021-07-14)
cargo --version
cargo 1.55.0-nightly (3ebb5f15a 2021-07-02)
Please help me with this.

sirhcel, 2021-07-15:
Hello @NitinSaxenait, if I remember correctly, the crate name changed a while ago. The crate to go for is `itm` now and re-installing it via worked as expected:
```shell
$ cargo install --force itm
    Updating crates.io index
  Installing itm v0.3.1
[...]
    Replaced package `itm v0.3.1` with `itm v0.3.1` (executable `itmdump`)
```
I'm doing this on stable which is at version 1.53 as of now. How are you attemting the installation?

NitinSaxenait, 2021-07-15:
@sirhcel i am still getting the same error 
`knoldus@knoldus-Vostro-3590:~$ cargo install --force itm
    Updating crates.io index
error: there is nothing to install in `itm v0.3.1`, because it has no binaries
`cargo install` is only for installing programs, and can't be used with libraries.
To use a library crate, add it as a dependency in a Cargo project instead.`
rustc --version
rustc 1.55.0-nightly (2f391da2e 2021-07-14)

sirhcel, 2021-07-15:
@NitinSaxenait this leaves me clueless for the moment. Could you give it a try on stable?

NitinSaxenait, 2021-07-15:
@sirhcel sorry i am not getting it. What you want me to do as "stable"?

sirhcel, 2021-07-15:
> @sirhcel sorry i am not getting it. What you want me to do as "stable"?
> okay I got it what you meant and I am getting this after update my stable version to 1.53. 
 
Exactly that. :)

> Still cargo install itm is not working.

I still have no explanation why. A while ago `itmdump` moved to https://github.com/japaric/itm-tools but I don't see how this affects the latest released version of `itm` where the [tool is present in 0.3.1](https://github.com/rust-embedded/itm/tree/v0.3.1/src/bin). I purged my cargo registry so that everything got downloaded and built again and installing `itmdump` still works as expected:
```shell
$ cargo uninstall itm
    Removing /[...]/.cargo/bin/itmdump
$ rm -rf ~/.cargo/registry
$ cargo install itm
    Updating crates.io index
  Downloaded itm v0.3.1
  Downloaded 1 crate (15.6 KB) in 1.27s
  Installing itm v0.3.1
  Downloaded log v0.3.9
[...]
    Finished release [optimized] target(s) in 49.37s
  Installing /[...]/.cargo/bin/itmdump
   Installed package `itm v0.3.1` (executable `itmdump`)
$ which itmdump
/[...]/.cargo/bin/itmdump
$ itmdump --version
itmdump 0.3.1
```

I'm out of ideas for the moment. Could you try my procedure from above as an act of desperation?

NitinSaxenait, 2021-07-15:
@sirhcel Yes the second command `rm -rf ~/.cargo/registry`  finally worked for me. Thanks :)
Can I know what I have done using this command?

sirhcel, 2021-07-15:
> @sirhcel Yes the second command `rm -rf ~/.cargo/registry` finally worked for me. Thanks :)

@NitinSaxenait, I'm glad to hear that cleaning the registry solved your issue. I assume that the `itmdump` binary is now available on your system and you could run it.

> Can I know what I have done using this command?

To my understanding this wipes Cargo's download cache and some meta information. This causes Cargo to download the crate sources on the next build. May be it was corrupted somehow and had a bad reference to `itm`. At least it looks like your issue is fixed now.

While looking into Cargo's cache, I found [cargo-cache](https://github.com/matthiaskrgr/cargo-cache) which allows to clean up registry and stuff without risking an `rm -rf` to the wrong place.

NitinSaxenait, 2021-07-16:
Ohh okay now i can work in peace :) Thanks for the help @sirhcel

closed 2021-07-16 by NitinSaxenait