rust-embedded/discovery #379: Multiple variations on cargo config

anthonyjmartinez, 2021-07-18:
Prior to ch. 7 on Registers, each chapter by default uses the overall project's default `src/.cargo/config.toml`, which some chapters suggest modifying to use your platform's specific `gdb`.

This is overridden for the first time in Ch 7 by an unmentioned `src/07-registers/.cargo/config` with a default assumption that `arm-none-eabi-gdb` is to be used. In my case, no such `gdb` exists, and `cargo run` fails. Editing as follows allowed me to continue, but some may not understand the resulting error from Cargo well enough (or realize that you can have a multitude of .cargo/config or .cargo/config.toml files) to fix the issue:

```
diff --git a/src/07-registers/.cargo/config b/src/07-registers/.cargo/config
index f5a1d95..e11ff2f 100644
--- a/src/07-registers/.cargo/config
+++ b/src/07-registers/.cargo/config
@@ -1,5 +1,5 @@
 [target.thumbv7em-none-eabihf]
-runner = "arm-none-eabi-gdb -q -x openocd.gdb"
+runner = "gdb -q -x openocd.gdb"
 #rustflags = [
 #  "-C", "link-arg=-Tlink.x",
 #]
```

Maybe it's worth adding a helpful note in chapters where a custom config exists so users know exactly what to modify to get back to a working state?

```
[user@projects discovery]$ find . -type f -wholename "*/.cargo/config*" | sort
./src/.cargo/config.toml
./src/07-registers/.cargo/config
./src/08-leds-again/.cargo/config
./src/09-clocks-and-timers/.cargo/config
./src/11-usart/.cargo/config
./src/14-i2c/.cargo/config
./src/15-led-compass/.cargo/config
./src/16-punch-o-meter/.cargo/config
./src/WIP-async-io-the-future/.cargo/config
```

eldruin, 2021-07-18:
Hmm, I guess those are left overs not removed in #308.
@winksaville is there any reason why those other config files should stay?

yoloVoe, 2023-01-04:
I deleted the `.cargo` directory under `f3discovery/src/07-registers`, and now `cargo run` is successfully using the appropriate config from `f3discovery/src/.cargo`.