rust-embedded/discovery #419: GDB: Os { code: 98, kind: AddrInUse, message: "Address already in use" }

Raunak-Singh-Inventor, 2021-12-22:
In Chapter 5.2 Flash It, when I run `cargo embed --features v2 --target thumbv7em-none-eabihf`, I get this error:
```
    Finished dev [unoptimized + debuginfo] target(s) in 0.01s
      Config default
      Target /home/raunaksingh/RustProjects/discovery/microbit/target/thumbv7em-none-eabihf/debug/led-roulette
        WARN probe_rs::config::registry > Found chip nRF52833_xxAA which matches given partial name nrf52833. Consider specifying its full name.
     Erasing sectors ✔ [00:00:00] [###################################################################]  4.00KiB/ 4.00KiB @ 32.80KiB/s (eta 0s )
 Programming pages   ✔ [00:00:00] [###################################################################]  4.00KiB/ 4.00KiB @ 17.86KiB/s (eta 0s )
    Finished flashing in 0.283s
During the execution of GDB an error was encountered:
Os { code: 98, kind: AddrInUse, message: "Address already in use" }
        Done processing config default
```

The code seems to build and flash successfully, but the command exits with the error `Address already in use`.

Help would be appreciated.

adamgreig, 2021-12-22:
Do you have another gdb session running somewhere, for example in another terminal or in the background? The error probably means a port is already in use. I think by default probe-rs runs its gdb stub on port 1337, but you can change this in your `Embed.toml` file:

```toml
[default.gdb]
enabled = true
gdb_connection_string = "127.0.0.1:1337"
```

Raunak-Singh-Inventor, 2021-12-22:
adding the `gdb_connection_string` to my `Embed.toml` fixed the error. Thanks.

closed 2021-12-22 by Raunak-Singh-Inventor