rust-embedded/discovery #441: (mis)Optimization can't run release version

gravit22, 2022-02-06:
When I try to run release version I can't even enter the main function with "step", although I can run unoptimized version completely. 
`Reading symbols from target/thumbv7em-none-eabihf/release/registers...
0x08000252 in registers::__cortex_m_rt_main ()
Loading section .vector_table, size 0x194 lma 0x8000000
Loading section .text, size 0x10fc lma 0x8000194
Loading section .rodata, size 0x1a0 lma 0x8001290
Start address 0x08000194, load size 5168
Transfer rate: 11 KB/sec, 1722 bytes/write.
Breakpoint 1 at 0x8000224
Note: automatically using hardware breakpoints for read-only addresses.
Breakpoint 2 at 0x80004cc
Breakpoint 3 at 0x800128c

Breakpoint 1, 0x08000224 in main ()
(gdb) s
Single stepping until exit from function main,
which has no line number information.
halted: PC: 0x0800022a

`

My code:
`#![no_main]
#![no_std]

#[allow(unused_imports)]
use aux7::{entry, iprintln, ITM, RegisterBlock};
use core::ptr;

#[entry]
fn main() -> ! {
    let gpioe = aux7::init().1;

    unsafe {
        // A magic address!
        const GPIOE_BSRR: u32 = 0x48001018;

        // Turn on the "North" LED (red)
        ptr::write_volatile(GPIOE_BSRR as *mut u32, 1 << 9);

        // Turn on the "East" LED (green)
        ptr::write_volatile(GPIOE_BSRR as *mut u32, 1 << 11);

        // Turn off the "North" LED
        ptr::write_volatile(GPIOE_BSRR as *mut u32, 1 << (9 + 16));

        // Turn off the "East" LED
        ptr::write_volatile(GPIOE_BSRR as *mut u32, 1 << (11 + 16));
    }

    loop {}
}`

sirhcel, 2022-02-06:
As far as I'm overseeing your description, this looks like the expected behaviour of enabling optimizations. To my understanding, optimizing the program execution is about to "break up" the lines you've written an get to the result as fast as possible. That's why there is typically no source line stepping in an optimized version of a program and you could consider lucky if function boundaries still exist. Your breakpoint at `main()` works and so this looks good to me at a first glance.

What about turning on a LED (or a pattern) at the end so that you could check at least the final result? Running the code above, I would expect at best to see a faint flicker.

gravit22, 2022-02-06:
Oh.... Yea, you are right. The program works correctly but debugging doesent. I just expected to debug it like in the tutorial, but I guess it is fine. 
Thank you for help.

closed 2022-02-06 by gravit22

sirhcel, 2022-02-06:
Glad to read that. And I bet debugging just does not meet your expectations despite being absolutely correct. You could call `disassemble` once you hit the breakpoint for `main` and you should get a pretty short assembly listing containing writes to all the locations you specified beforehand.