rust-embedded/discovery #451: Unresolved imports disables syntax highlighting and code completion

kC0pter, 2022-04-26:
First of all: Really awesome work with the two books and the practical examples.
I am currently doing the examples with the microbit:v2 and i have suggestion.

Since the book can be done with both the v1 and the v2 of the board, you can either enable v1 or v2 via features during compilation.
In VSCode with Rust Analyzer that becomes a problem since there is no default feature which makes the analyzer not include some modules and therefore throwing a lot of unresolved imports errors.
My idea is to either add the v2 as a default feature since it is the newer board or to split the examples to different branches.
Then one can choose which branch to check out and only work with that boards examples (this would also get rid of the double definition code with #[cfg(feature = ...)] above it).

PS: Git dependencies can also be an issue with the analyzer since sometimes they cannot be resolved properly and produce errors as well. This is a known problem apparently.

doehyunbaek, 2022-05-11:
Hi. Did you try setting Vscode rust-analyzer.cargo.features option? It's nor perfect but helps quite a bit with packages containing feature flags. It's possible to use this per-workspace basis. Not sure about different feature flags inside subdirectories. But this helped a bunch for me.

[image]

jmevel, 2022-05-17:
> Hi. Did you try setting Vscode rust-analyzer.cargo.features option? It's nor perfect but helps quite a bit with packages containing feature flags. It's possible to use this per-workspace basis. Not sure about different feature flags inside subdirectories. But this helped a bunch for me.
> 
> [image]

Thanks for this tips. However while it does indeed recognize the crates after setting `"rust-analyzer.cargo.features": "all"` it has the downside of not showing any error nor warning anymore. One can only know if the code is correct by running a `cargo build`.

So now I'm constantly enabling and disabling this feature and relaunching VS Code...
Is there a way to make rust-analyzer keep showing us errors and warnings even with this setting enabled?

Thanks for your answer

jmevel, 2022-05-17:
@doehyunbaek Actually re-checking your answer it seems you've a setting that I don't have. I don't see any way I can add several custom items like `v2`. All I can do is to set `"rust-analyzer.cargo.features": "all"`

[image]

Your comment is only 6 days old so I doubt that's a deprecated feature. Is it a beta feature or something like this? 
How exactly could we access the same thing as on your screenshot?

Thanks

doehyunbaek, 2022-05-17:
Hi. It seems there was regression or intented change between v0.2.1048 and v0.3.1051. The screenshot was taken at v0.2.1048 versionn and as I updated the extension to v0.3.1051, it also accepts only "all" as value.

doehyunbaek, 2022-05-19:
Seems to be fixed in v0.3.1059! I think https://github.com/rust-lang/rust-analyzer/pull/12229 fixed the issue.

eldruin, 2022-07-05:
Closing since it seems fixed.

closed 2022-07-05 by eldruin