rust-embedded/discovery #452: [src/05-led-roulette]No loadable ELF sections were found

Kotodian, 2022-05-17:
[image]

sirhcel, 2022-07-06:
Strange thing. This works out-of-the-box for me with 3c7bcec and Rust 1.62.0. I've modifed `Embed.toml` for the micro:bit v2
```diff
discovery/microbit/src/05-led-roulette $ git diff
diff --git a/microbit/src/05-led-roulette/Embed.toml b/microbit/src/05-led-roulette/Embed.toml
index 22375a6..d07a386 100644
--- a/microbit/src/05-led-roulette/Embed.toml
+++ b/microbit/src/05-led-roulette/Embed.toml
@@ -1,5 +1,5 @@
 [default.general]
-# chip = "nrf52833_xxAA" # uncomment this line for micro:bit V2
+chip = "nrf52833_xxAA" # uncomment this line for micro:bit V2
 # chip = "nrf51822_xxAA" # uncomment this line for micro:bit V1

 [default.reset]
```
and could qsuccessfully flash with `cargo-embed`:
```console
discovery/microbit/src/05-led-roulette $ cargo embed --features v2 --target thumbv7em-none-eabihf
    Finished dev [unoptimized + debuginfo] target(s) in 0.04s
      Config default
      Target /Users/christian/Entwicklung/rust/discovery/microbit/target/thumbv7em-none-eabihf/debug/led-roulette
     Erasing sectors ✔ [00:00:00] [################################################################################################]  4.00KiB/ 4.00KiB @ 31.42KiB/s (eta 0s )
 Programming pages   ✔ [00:00:00] [################################################################################################]  4.00KiB/ 4.00KiB @ 16.67KiB/s (eta 0s )
    Finished flashing in 0.303s
```

There is nothing interesting to see with the default code. My slightly modified solution [`examples/my-solution.rs](https://github.com/sirhcel/discovery/blob/issue-452-no-loadable-elf-sections/microbit/src/05-led-roulette/examples/my-solution.rs) makes the light go round after flashing it like
```console
discovery/microbit/src/05-led-roulette $ cargo embed --features v2 --target thumbv7em-none-eabihf --example my-solution
```
and pressing the reset button afterwards.

Jzow, 2022-09-25:
I also had this problem at first. Then I plugged in the USB again and rebuilt the cargo clean according to the steps

XaydBayeck, 2022-10-18:
I copy the `.cargo/config` file to my own project. And it's work... 
```shell
discovery-my on  master [?] is 📦 v0.1.0 via 🦀 v1.66.0-nightly
❯ cat .cargo/config
───────┬─────────────────────────────────────────────────
       │ File: .cargo/config
───────┼─────────────────────────────────────────────────
   1   │ [target.'cfg(all(target_arch = "arm", target_os = "none"))']
   2   │ rustflags = [
   3   │   "-C", "link-arg=-Tlink.x",
   4   │ ]
───────┴─────────────────────────────────────────────────
```

If you compare with or without this file, you will find the `Entry point address` is different by running `cargo readobj`.

Without this config file, the `Entry point address` is `0x0`. I guess this is the source of problem.

l-0-l, 2024-06-12:
Yes, the reply mentioned above solves the problem. The reason is that the crate `cortex-m-rt`, the basic crate needed to work with this family of controllers, copies a default `link.x` file (you can search for `link.x.in` in the crate's source code) with Linker instructions to the build directory, and we must instruct the build system to use it. Furthermore, `link.x` file references a file called `memory.x`, which must be present and located somewhere on Linker's search path (`rustc-link-search`). This file should contain at least a minimal setting for the flash memory and the RAM on the target device, like:
```
MEMORY
{
  FLASH : ORIGIN = 0x00000000, LENGTH = 256K
  RAM : ORIGIN = 0x20000000, LENGTH = 16K
}
```
Further reading:
- [cortex-m-rt documentation](https://docs.rs/cortex-m-rt/latest/cortex_m_rt/)
- [build scripts](https://doc.rust-lang.org/cargo/reference/build-scripts.html)
- [lingade](https://doc.rust-lang.org/reference/linkage.html)