rust-embedded/discovery #453: microbit = "^0.12.0" does not match 0.13.0

jim2236, 2022-05-29:
I started working with discovery book and have a microbit v2.20.  I working on a Mac M1 running Monterey 12.4.  I worked through all setup steps in the book up to Verify cargo-embed.  I repeated all steps up to that point to ensure I did followed instructions correctly, including rustc, cargo-binutils, cargo-embed, the ARM GCC toolchain and Minicom through Homebrew, modifying the Embed.toml file to uncomment the line for the V2 chip, and ran rustup target add thumbv7em-none-eabihf successfully.  I cloned theGithub repo https://github.com/rust-embedded/discovery/

I also flashed the firmware on the Microbit to ensure it is current.  The Interface Version in the Details.txt file is 0256. Here is the contents of Details.txt

# DAPLink Firmware - see https://daplink.io
Build ID: alpha9-189-g5dd23001 (gcc)
Unique ID: 99053602000528336f6efa0a9a18264c000000006e052820
HIC ID: 6e052820
Auto Reset: 1
Automation allowed: 0
Overflow detection: 0
Incompatible image detection: 1
Page erasing: 0
Daplink Mode: Interface
Interface Version: 0256
Bootloader Version: 0256
Git SHA: 5dd23001a7a3199d74870790049d6686e183316c
Local Mods: 0
USB Interfaces: MSD, CDC, HID, WebUSB
Bootloader CRC: 0xa60a7780
Interface CRC: 0x0bac75fa
Remount count: 0
URL: https://microbit.org/device/?id=9905&v=0256

I'm working in the /discovery/microbit/src subdirectory on my Mac. 

When I run cargo embed --target thumbv7em-none-eabihf, it return the following result / error.

03-setup % cargo embed --target thumbv7em-none-eabihf
    Updating crates.io index
    Updating git repository `https://github.com/nrf-rs/microbit/`
error: failed to select a version for the requirement `microbit = "^0.12.0"`
candidate versions found which didn't match: 0.13.0
location searched: Git repository https://github.com/nrf-rs/microbit/
required by package `led-roulette v0.1.0 (/Users/squeeks/rust/projects/microbit/discovery/microbit/src/05-led-roulette)`
       Error Failed to run cargo build: exit code = Some(101).

Cargo appears to be looking for a different microbit version number. 

Any help to resolve the issue would be appreciated.

fawohlsc, 2022-05-29:
@jim2236 - Ran into the same issues. Within the 'Cargo.toml' files located in the 'src' subfolders, change the version of microbit from '0.12.0'to '0.13.0'. Hope that helps.

jim2236, 2022-05-29:
The solution worked.  Thank you!

closed 2022-05-29 by jim2236