rust-embedded/discovery #464: Error when using re-running the cargo embed to flash a program

videbar, 2022-09-03:
I'm following the examples in the rust embedded book and I have encountered a very bizarre issue when flashing programs. I'm using the microbit v2 board on Linux (Fedora 36) and I'm currently working with the LED roulette example.

I have set up the configuration files according to the book and I'm able to flash programs using the command:

`cargo embed --features v2 --target thumbv7em-none-eabihf`

However, this only works if I type the command (or paste it on the terminal), if I press the up key on my keyboard to execute it again, or run it from the command history, I'll get the following error (in this case I'm running it from the history):

```
$ !3006
cargo embed --features v2 --target humbv7em-none-eabihf
error: failed to run `rustc` to learn about target-specific information

Caused by:
  process didn't exit successfully: `rustc - --crate-name ___ --print=file-names --target humbv7em-none-eabihf --crate-type bin --crate-type rlib --crate-type dylib --crate-type cdylib --crate-type staticlib --crate-type proc-macro --print=sysroot --print=cfg` (exit status: 1)
  --- stderr
  error: Error loading target specification: Could not find specification for target "humbv7em-none-eabihf". Run `rustc --print target-list` for a list of built-in targets

       Error Failed to run cargo build: exit code = Some(101).
```
Again, this doesn't happen if I type the command on the terminal, only if I re-run a previous copy of the command.

adamgreig, 2022-09-03:
In your failing example, the target name is mis-spelled as `humbv7em-none-eabihf` (missing initial `t`), which is causing the error you're getting. Could it be you're running a bad command from history or your shell's history replacement is somehow going wrong?

videbar, 2022-09-04:
That's exactly the problem, I feel quite dumb for not noticing. Thanks for pointing it out!

closed 2022-09-04 by videbar