rust-embedded/discovery #468: Can't follow the book because gdb isn't available on Apple M1

thehappycoder, 2022-09-25:
This could be a good alternative https://github.com/knurling-rs/probe-run

aronzvi, 2022-10-12:
@thehappycoder 
Can you please provide details on how to use probe-run instead of gdb?

lpython, 2022-11-06:
I've gotten halfway through the book on an M1 air, what exactly didn't work? So far, its been better than any attempt I've made on windows. I installed gdb through homebrew.

slyons, 2022-11-28:
@thehappycoder I was able to run it after installing all of the targets by using `arm-none-eabi-gdb` instead of `gdb`

rashack, 2023-01-14:
@slyons can you please elaborate on what you did to make it work? What did you install and how? And what did you use instead of gdb?

lpython, 2023-01-15:
@rashack  nothing too special, I've only have been using a Mac for a year and use brew to install most cli programs.

brew install arm-none-eabi-gcc
brew install minicom openocd

I've used the previous [rust embedded discovery book](https://docs.rust-embedded.org/discovery/f3discovery/index.htmlurl) which really is still current, just for a different board. 
Specifically, this [page](https://docs.rust-embedded.org/discovery/f3discovery/03-setup/index.html) should cover all the details

a file located at ./src/.cargo/config.toml should identify the target for the executable and the command to run which is something like
bash```
arm-none-eabi-gdb -q -x ../openocd.gdb
```

rashack, 2023-01-15:
@slyons Ah, thank you! What did it for me was `brew install arm-none-eabi-gcc` and then I got it going with `.../discovery/microbit $ arm-none-eabi-gdb target/thumbv7em-none-eabihf/debug/led-roulette`

boathouse2112, 2023-07-10:
Seems like arm-none-eabi-gcc is no longer listed on homebrew. Had to tap https://github.com/eblot/homebrew-armeabi to get it.