rust-embedded/discovery #469: How should I run it? My microbit v2 device has no led light

Jzow, 2022-09-26:
log:

```rust
➜  discovery git:(gh-pages) ✗ cd microbit/05-led-roulette 
➜  05-led-roulette git:(gh-pages) ✗ cargo embed --features v2 --target thumbv7em-none-eabihf
    Finished dev [unoptimized + debuginfo] target(s) in 0.31s
      Config default
      Target /Users/jameszow/Documents/RustWorkSpace/discovery/microbit/05-led-roulette/target/thumbv7em-none-eabihf/debug/led-roulette
     Erasing sectors ✔ [00:00:00] [#######################################################################################]  4.00KiB/ 4.00KiB @ 26.19KiB/s (eta 0s )
 Programming pages   ✔ [00:00:00] [#######################################################################################]  4.00KiB/ 4.00KiB @ 11.60KiB/s (eta 0s )
    Finished flashing in 0.419s
```

I didn't see the effect

Jzow, 2022-09-26:
By the way, what is the command to enter gdb? I used this command, but it is obviously incorrect

```shell
arm-none-eabi-gdb -q -ex "target remote :1337" target/thumbv7em-none-eabihf/debug/led-roulette
```

mattjperez, 2022-09-30:
did you uncomment the line in `Embed.toml` like we had to do in `03-setup`?

As for gdb, from the `discovery/microbit` directory, I used `gdb-multiarch target/thumbv7em-none-eabihf/debug/led-roulette` followed by `(gdb) target remote :1337`