rust-embedded/discovery #472: [src/05-led-roulette] microbit v2.2 "Error attaching to RTT"

mattjperez, 2022-10-02:
TLDR: microbit v2.2 runs then fails immediately when using RTT.

Section 5.5 requires us to use the `rtt-target` crate however it appears the microbit v2.2 is having issues regarding RTT.
I believe this is related to https://github.com/probe-rs/probe-rs/issues/1235.

```
Finished flashing in 1.964s
    GDB stub listening at 127.0.0.1:1337
       Error Error attaching to RTT: RTT control block not found in target memory.
             - Make sure RTT is initialized on the target, AND that there are NO target breakpoints before RTT initalization.
             - For VSCode and probe-rs-debugger users, using `halt_after_reset:true` in your `launch.json` file will prevent RTT
                initialization from happening on time.
             - Depending on the target, sleep modes can interfere with RTT.
```

To confirm, I removed the call to `rprintln!()` and replaced `rtt_init_print` with `rtt_init_default`, and received the same results as above.
 
```toml
# Embed.toml
[default.general]
chip = "nrf52833_xxAA" # uncomment this line for micro:bit V2

[default.reset]
halt_afterwards = true

[default.rtt]
enabled = true

[default.gdb]
enabled = true
```

closed 2022-10-02 by mattjperez

vincentserpoul, 2023-07-30:
@mattjperez I am seeing the same error, what did you do to go past this?
(I tried multiple config of embed.toml, but i still get an error.

glennpjones, 2024-02-14:
> @mattjperez I am seeing the same error, what did you do to go past this? (I tried multiple config of embed.toml, but i still get an error.

setting `halt_afterwards = false` fixed this for me