rust-embedded/discovery #474: [src/07-registers] not able to run

a-gradina, 2022-10-10:
Up until this chapter I was able to code along. cargo run throws an error for this chapter and onward.

```test
agradina@pop-os:~/discovery/f3discovery/src/07-registers$ cargo run
    Finished dev [unoptimized + debuginfo] target(s) in 0.01s
     Running `arm-none-eabi-gdb -q -x openocd.gdb /home/agradina/discovery/f3discovery/target/thumbv7em-none-eabihf/debug/registers`
error: could not execute process `arm-none-eabi-gdb -q -x openocd.gdb /home/agradina/discovery/f3discovery/target/thumbv7em-none-eabihf/debug/registers` (never executed)

Caused by:
  No such file or directory (os error 2)

```

I have GDB set to gdb-multiarch -q in the config.toml file.

BikiniBottomsVeryOwn, 2022-10-12:
Also experiencing this exact same error. Just came here to post this issue.

[image]

BikiniBottomsVeryOwn, 2022-10-12:
Also was getting this exact same error when running "cargo run" on chapters 8 and 9. Didn't check any further but this problem may span the rest of the chapters.

yoloVoe, 2023-01-04:
This issue is a duplicate of another older issue: https://github.com/rust-embedded/discovery/issues/379

We should resolve this.

closed 2024-10-24 by a-gradina