rust-embedded/discovery #485: Unclear sentence in 06-hello-world

ConnorMooneyhan, 2022-11-20:
The following sentence in chapter 6 just doesn't really make sense: "Also, it setup break points and stepped through the trampoline we are at the first executable statement in `fn main()`." I'm not quite sure what a better wording would be because I can't tell exactly what the author is trying to say, but perhaps it's something like "It also set up breakpoints and stepped through the [trampoline](https://stackoverflow.com/questions/189725/what-is-a-trampoline-function), so now we are at the first executable statement in `fn main()`." I think it's important to include a link with the word "trampoline" since it's the first time that's used in this text and it's likely to be new to the reader; it certainly was to me. The Wikipedia page was sparse and high-level, so I chose to link to a StackOverflow answer in my suggestion, but another explanation could be chosen.

eldruin, 2022-11-25:
Sounds like a good improvement! I would certainly welcome this as a PR

Jzow, 2022-12-03:
I also think there is a should better explanation for the "trampoline"