rust-embedded/discovery #487: Some images should be stored locally and imported in html page

Jzow, 2022-12-03:
**Question**:

For the gh-pages branch, some images are imported from Wikipedia. However, when we translate the book, some images cannot be loaded successfully in some regions due to vpn.

**Example**:

Picture of `index.html` in Chapter 10

**Idea**:

This is my personal idea, not everything. We can create an image folder under each chapter to store the downloaded images. The html page can be imported through the relative path.

I hope it will be better for users to read this book

eldruin, 2022-12-06:
I see the problem. However, I am unsure about the copyright issues since each image has a specific license, author and so on.
Any idea how to solve this?

Jzow, 2022-12-07:
My previous practice was to indicate the source of the picture, except for the pictures published by Wikipedia.

The 《discovery book》 was not used for commercial purposes, and it will not damage the author's reputation, property and rights.

But he may have legal problems. I think we can pause until there is a better idea.