rust-embedded/discovery #490: hidapi error: Permission denied

greenfierydragon, 2022-12-04:
When I run the following command:

```
$ cargo embed --features v1 --target thumbv6m-none-eabi
```

I get:

```
       Error Probe could not be created

             Caused by:
                 0: hidapi error: Permission denied
                 1: hidapi error: Permission denied
```

Despite having:

```
$ cat /etc/udev/rules.d/99-microbit.rules

# CMSIS-DAP for microbit
SUBSYSTEM=="usb", ATTR{idVendor}=="0d28", ATTR{idProduct}=="0204", MODE:="666"
```

I have rebooted several times and even checked the file permission on my device (`/dev/bus/usb/001/010`):

```
$ ls /dev/bus/usb/001/010
crw-rw-rw- 189,9 root  4 Dec 07:17 /dev/bus/usb/001/010
```

My kernel is 6.0.10 and I run archlinux. Any help would be much apreciated.

sudoBash418, 2022-12-07:
Not sure if this will fix your issue, but I was facing the same error and fixed it with the following udev rule:
```
# CMSIS-DAP for microbit
SUBSYSTEM=="usb", ATTRS{idVendor}=="0d28", ATTRS{idProduct}=="0204", MODE="0660", TAG+="uaccess"
```

According to the [Arch Wiki](https://wiki.archlinux.org/title/Udev#Allowing_regular_users_to_use_devices):
> The modern recommended approach for systemd systems is to use a `MODE` of `660` to let the group use the device, and then attach a TAG named `uaccess`.

After updating the udev rule, I ran `sudo udevadm control --reload; sudo udevadm trigger` and was then able to run `cargo embed --target thumbv6m-none-eabi` without issue.

Vortoman, 2022-12-15:
Had the exact same problem, solution worked. Thanks a lot!

eldruin, 2023-01-16:
Nice to see that you found a solution.
Could one of you put together a PR adding this to the book?

bcpeinhardt, 2023-02-13:
Same fix worked on Ubuntu, with a `udevadm trigger` instead of replugging in. :)

l-0-l, 2024-06-11:
Didn't work for me with a v1, not sure why. On Debian 12. Provided my solution [here](https://github.com/rust-embedded/discovery/issues/550).

BartMassey, 2024-06-12:
@l-0-l Thanks for that.

Note that we are currently planning to remove Microbit v1 from this guide to reduce confusion. If folks feel strongly that we need to continue to support this board, please comment on https://github.com/rust-embedded/wg/pull/759