rust-embedded/discovery #502: Failing to verify the installation in section 3.4

rashack, 2023-01-14:
I have a microbit v.2.21, connected to a recently installed Debian 11.6
Here is the error:
```
$ cargo embed --target thumbv7em-none-eabihf
    Finished dev [unoptimized + debuginfo] target(s) in 0.01s
      Config default
      Target /home/kjell/src/rust/discovery/microbit/target/thumbv7em-none-eabihf/debug/rtt-check
        WARN probe_rs::architecture::arm::memory::romtable > Component at 0xe0001000: CIDR0 has invalid preamble (expected 0xd, got 0x0)
        WARN probe_rs::architecture::arm::memory::romtable > Component at 0xe0001000: CIDR2 has invalid preamble (expected 0x5, got 0x0)
        WARN probe_rs::architecture::arm::memory::romtable > Component at 0xe0001000: CIDR3 has invalid preamble (expected 0xb1, got 0x0)
        WARN probe_rs::architecture::arm::memory::romtable > Component at 0xe0000000: CIDR0 has invalid preamble (expected 0xd, got 0xb1)
        WARN probe_rs::architecture::arm::memory::romtable > Component at 0xe0000000: CIDR1 has invalid preamble (expected 0x0, got 0x1)
        WARN probe_rs::architecture::arm::memory::romtable > Component at 0xe0000000: CIDR2 has invalid preamble (expected 0x5, got 0xb1)
        WARN probe_rs::architecture::arm::memory::romtable > Component at 0xe0040000: CIDR0 has invalid preamble (expected 0xd, got 0x0)
        WARN probe_rs::architecture::arm::memory::romtable > Component at 0xe0040000: CIDR2 has invalid preamble (expected 0x5, got 0x0)
        WARN probe_rs::architecture::arm::memory::romtable > Component at 0xe0040000: CIDR3 has invalid preamble (expected 0xb1, got 0x0)
       Error failed attaching to target

             Caused by:
                 0: Unable to load specification for chip
                 1: The connected chip could not automatically be determined.
```
Some information about my machine:
```
$ cat /etc/debian_version
11.6

$ pwd
.../discovery/microbit/src/03-setup

$ rustc -V
rustc 1.66.1 (90743e729 2023-01-10)

$ cargo size --version
cargo-size 0.3.6

$ cargo embed --version
cargo embed 0.14.0
git commit: crates.io

$ git --no-pager diff
diff --git a/microbit/src/05-led-roulette/Embed.toml b/microbit/src/05-led-roulette/Embed.toml
index 22375a6..d07a386 100644
--- a/microbit/src/05-led-roulette/Embed.toml
+++ b/microbit/src/05-led-roulette/Embed.toml
@@ -1,5 +1,5 @@
 [default.general]
-# chip = "nrf52833_xxAA" # uncomment this line for micro:bit V2
+chip = "nrf52833_xxAA" # uncomment this line for micro:bit V2
 # chip = "nrf51822_xxAA" # uncomment this line for micro:bit V1

 [default.reset]

$ cat /etc/udev/rules.d/99-microbit.rules
# CMSIS-DAP for microbit
SUBSYSTEM=="usb", ATTR{idVendor}=="0d28", ATTR{idProduct}=="0204", MODE:="666"
```

dunmengjun, 2023-03-10:
@rashack It may be that you forgot uncomment below code in Embed.toml
`
[default.general]
chip = "nrf52833_xxAA" # uncomment this line for micro:bit V2
# chip = "nrf51822_xxAA" # uncomment this line for micro:bit V1
`

protofarer, 2023-05-25:
Worked for me. Thanks @dunmengjun