rust-embedded/discovery #505: cargo embed fails with "no external crate `microbit`"

jamesbebbington, 2023-01-21:
Attempting to get set up with a micro:bit v2.21:

```
$ rustup target add thumbv7em-none-eabihf
info: downloading component 'rust-std' for 'thumbv7em-none-eabihf'
info: installing component 'rust-std' for 'thumbv7em-none-eabihf'

$ cargo embed --target thumbv7em-none-eabihf
   Compiling led-roulette v0.1.0 (/Users/james/Developer/rust/discovery/microbit/src/05-led-roulette)
   Compiling panic-rtt-target v0.1.2
error: could not compile `led-roulette` due to 2 previous errors
warning: build failed, waiting for other jobs to finish...
error[E0432]: unresolved import `microbit`
 --> src/05-led-roulette/src/main.rs:7:5
  |
7 | use microbit as _;
  |     ^^^^^^^^^^^^^ no external crate `microbit`

error: aborting due to previous error

       Error Failed to run cargo build: exit code = Some(101).
```

Deets:

```
$ sw_vers
ProductName:	macOS
ProductVersion:	12.6.2
BuildVersion:	21G320

$ rustc -V
rustc 1.66.0 (69f9c33d7 2022-12-12)

$ cargo size --version
cargo-size 0.3.3

$ cargo embed --version
cargo-embed 0.11.0
git commit: crates.io

$ git diff
diff --git a/microbit/src/03-setup/Embed.toml b/microbit/src/03-setup/Embed.toml
index d84d0ac..6b5accb 100644
--- a/microbit/src/03-setup/Embed.toml
+++ b/microbit/src/03-setup/Embed.toml
@@ -2,7 +2,7 @@
 protocol = "Swd"

 [default.general]
-# chip = "nrf52833_xxAA" # uncomment this line for micro:bit V2
+chip = "nrf52833_xxAA" # uncomment this line for micro:bit V2
 # chip = "nrf51822_xxAA" # uncomment this line for micro:bit V1

 [default.rtt]
```

jamesbebbington, 2023-01-21:
Tried with latest versions of cargo-binutils (v0.3.6) and cargo-embed (v0.14.2). Results in similar (but slightly different) sadness:

```
$ cargo embed --target thumbv7em-none-eabihf
   Compiling led-roulette v0.1.0 (/Users/james/Developer/rust/discovery/microbit/src/05-led-roulette)
   Compiling rtt-check v0.1.0 (/Users/james/Developer/rust/discovery/microbit/src/03-setup)
   Compiling led-compass v0.1.0 (/Users/james/Developer/rust/discovery/microbit/src/09-led-compass)
   Compiling punch-o-meter v0.1.0 (/Users/james/Developer/rust/discovery/microbit/src/10-punch-o-meter)
   Compiling uart v0.1.0 (/Users/james/Developer/rust/discovery/microbit/src/07-uart)
   Compiling i2c v0.1.0 (/Users/james/Developer/rust/discovery/microbit/src/08-i2c)
error: could not compile `led-roulette` due to 2 previous errors
warning: build failed, waiting for other jobs to finish...
error: could not compile `punch-o-meter` due to 2 previous errors
error: could not compile `uart` due to 4 previous errors
error: could not compile `i2c` due to 5 previous errors
error: could not compile `led-compass` due to 5 previous errors
error[E0433]: failed to resolve: use of undeclared crate or module `microbit`
 --> src/08-i2c/src/main.rs:9:5
  |
9 | use microbit::hal::prelude::*;
  |     ^^^^^^^^ use of undeclared crate or module `microbit`

error[E0433]: failed to resolve: use of undeclared crate or module `microbit`
  --> src/09-led-compass/src/calibration.rs:11:5
   |
11 | use microbit::display::blocking::Display;
   |     ^^^^^^^^ use of undeclared crate or module `microbit`

error[E0433]: failed to resolve: use of undeclared crate or module `microbit`
  --> src/09-led-compass/src/main.rs:13:5
   |
13 | use microbit::{display::blocking::Display, hal::Timer};
   |     ^^^^^^^^ use of undeclared crate or module `microbit`

error[E0432]: unresolved import `microbit`
 --> src/05-led-roulette/src/main.rs:7:5
  |
7 | use microbit as _;
  |     ^^^^^^^^^^^^^ no external crate `microbit`

error[E0433]: failed to resolve: use of undeclared crate or module `microbit`
  --> src/10-punch-o-meter/src/main.rs:12:17
   |
12 |     let board = microbit::Board::take().unwrap();
   |                 ^^^^^^^^ use of undeclared crate or module `microbit`

error[E0433]: failed to resolve: use of undeclared crate or module `microbit`
  --> src/07-uart/src/main.rs:30:17
   |
30 |     let board = microbit::Board::take().unwrap();
   |                 ^^^^^^^^ use of undeclared crate or module `microbit`

error[E0433]: failed to resolve: use of undeclared crate or module `microbit`
  --> src/08-i2c/src/main.rs:32:17
   |
32 |     let board = microbit::Board::take().unwrap();
   |                 ^^^^^^^^ use of undeclared crate or module `microbit`

error[E0425]: cannot find value `serial` in this scope
  --> src/07-uart/src/main.rs:53:16
   |
53 |     nb::block!(serial.write(b'X')).unwrap();
   |                ^^^^^^ not found in this scope

error[E0425]: cannot find value `serial` in this scope
  --> src/07-uart/src/main.rs:54:16
   |
54 |     nb::block!(serial.flush()).unwrap();
   |                ^^^^^^ not found in this scope

error[E0433]: failed to resolve: use of undeclared crate or module `microbit`
  --> src/09-led-compass/src/main.rs:26:17
   |
26 |     let board = microbit::Board::take().unwrap();
   |                 ^^^^^^^^ use of undeclared crate or module `microbit`

error[E0425]: cannot find value `i2c` in this scope
  --> src/09-led-compass/src/main.rs:37:46
   |
37 |     let mut sensor = Lsm303agr::new_with_i2c(i2c);
   |                                              ^^^ not found in this scope

error[E0425]: cannot find value `i2c` in this scope
  --> src/08-i2c/src/main.rs:45:5
   |
45 |     i2c.write_read(ACCELEROMETER_ADDR, &[ACCELEROMETER_ID_REG], &mut acc).unwrap();
   |     ^^^ not found in this scope

error[E0425]: cannot find value `i2c` in this scope
  --> src/08-i2c/src/main.rs:46:5
   |
46 |     i2c.write_read(MAGNETOMETER_ADDR, &[MAGNETOMETER_ID_REG], &mut mag).unwrap();
   |     ^^^ not found in this scope

error: aborting due to previous error

For more information about this error, try `rustc --explain E0432`.
error: aborting due to previous error

For more information about this error, try `rustc --explain E0433`.
error: aborting due to 3 previous errors

Some errors have detailed explanations: E0425, E0433.
For more information about an error, try `rustc --explain E0425`.
error: aborting due to 4 previous errors

Some errors have detailed explanations: E0425, E0433.
For more information about an error, try `rustc --explain E0425`.
error: aborting due to 4 previous errors

Some errors have detailed explanations: E0425, E0433.
For more information about an error, try `rustc --explain E0425`.
       Error Failed to run cargo build: exit code = Some(101).
```

sirhcel, 2023-01-21:
You're almost there: it looks like your command line just misses selecting one of the micro:bit variants. The following should do the trick for you:
```
$ cargo embed --features v2 --target thumbv7em-none-eabihf
```

jamesbebbington, 2023-01-22:
Many thanks for your help! Turns out I was being an idiot and not reading the instructions properly, I hadn't `cd`-ed into `microbit/src/03-setup` beforehand.

closed 2023-01-22 by jamesbebbington