rust-embedded/discovery #508: 05-led-roulette | cargo embed crashes on (gdb) target remote :1337 with During the execution of GDB an error was encountered: failed to fill whole buffer

bcpeinhardt, 2023-02-13:
05-led-roulette | cargo embed crashes on `(gdb) target remote :1337` with `During the execution of GDB an error was encountered: failed to fill whole buffer`. 
Coming at this with zero domain knowledge (never even used GDB) so could use some help. Running Ubuntu 22 and following along with Microbit v2.

bcpeinhardt, 2023-02-13:
Oh and on gdb's side of thing:
```
(gdb) target remote :1337
Remote debugging using :1337
warning: while parsing target description (at line 4): Target description specified unknown architecture "armv7"
warning: Could not load XML target description; ignoring
Truncated register 16 in remote 'g' packet

```

SHA256man, 2023-02-18:
> Oh and on gdb's side of thing:
>
> ```
> (gdb) target remote :1337
> Remote debugging using :1337
> warning: while parsing target description (at line 4): Target description specified unknown architecture [...]

Yes! This has to do with the compiler and gdb. gdb on your system works for probably x86 architecture but you are trying to run it on a program compiled and running on ARM. If you followed all the installing instructions in this tutorial so far, you should have a specific version of gdb installed. In your terminal do `ls /bin | grep "gdb"`. This should return you all gdb related binaries. For me on arch it is `arm-none-eabi-gdb` and on ubuntu it should be something like `gdb-arm-none-eabi`. If there is nothing of the sort, you should install it via apt-get. Then run `gdb-arm-none-eabi target/thumbv7em-none-eabihf/debug/led-roulette`, assuming that this is the correct binary for ubuntu. Hope that helps.

omar-hazem-1996, 2023-03-17:
I am using **Kubuntu** as my Linux distro with a **micro:bit 2.21**
I am now in `discovery/microbit/src/05-led-roulette/`
This is how i solved it:
1. Run `sudo apt install gdb-arm-none-eabi`
2. Run `cargo embed --features v2 --target thumbv7em-none-eabihf` and do not terminate the process. Also I noticed that `cargo-embed` is using too much CPU (nearly 25%) and i don't know why
3. Go back 2 directories to `microbit/` and run `gdb-multiarch target/thumbv7em-none-eabihf/debug/led-roulette`
4. Run `target remote :1337`

umatbro, 2023-05-19:
I had the same problem on Arch (Manjaro). I roughly followed the steps above, the arch package is: https://archlinux.org/packages/community/x86_64/arm-none-eabi-gdb/

```bash
sudo pacman -S arm-none-eabi-gdb
arm-none-eabi-gdb target/thumbv7em-none-eabihf/debug/led-roulette
```

smidkristian, 2023-12-28:
yeah, the key is to run `gdb-multiarch` instead of just `gdb`

zhenyakornilov, 2024-07-25:
I can additionaly add If someone working on Windows then the key is to run `arm-none-eabi-gdb` instead of `gdb`, which you should have installed as dependency and add to Path in 3.2 Chapter of the Discovery book