rust-embedded/discovery #511: Chapter 05: Cannot see source code when starting gdb

noiseOnTheNet, 2023-02-19:
I tried to execute the code in GDB on Manjaro.
Compilation completes successfully and gdb-multiarch is installed
When I launch the gdb it writes:

Reading symbols from target/thumbv6m-none-eabi/debug/led-roulette...

so I guess it should be working.
It connects with "target remote :1337" but looks like it does not find symbols

Remote debugging using :1337
0xffffffff in ?? ()

When I try to use "list src" TUI writes "[No source available]"
Any suggestion?

rjknet, 2023-03-09:
I am having the same problem (with microbit v2). Did anyone manage to solve the problem by any chance?

dunmengjun, 2023-03-10:
The led-roulette elf file is not in 05-led-roulette folder, it's in microbit floder
```
\discovery\microbit> arm-none-eabi-gdb target/thumbv7em-none-eabihf/debug/led-roulette
```

eldruin, 2023-03-30:
Fixed in #520

noiseOnTheNet, 2023-04-01:
I just downloaded the latest commit and still have the same problem.

Any ideas? I completely recompiled the binary by deleting the build directories

nocommentsgood, 2023-07-29:
Not sure about the noise you get after setting the debug target to port 1337, but I fixed this by using 'continue' after setting the breakpoint on main. After issuing 'continue' use 'layout src'.