rust-embedded/discovery #515: Chapter 5.3: `monitor reset` is not supported by `cargo-embed` >= 0.13.0

arr-ee, 2023-03-12:
Chapter 5 refers to `monitor reset` gdb command in the [`Debug it` section](https://github.com/rust-embedded/discovery/blob/7d879f301dc364b3858eff6fbb90514c50587e91/microbit/src/05-led-roulette/debug-it.md?plain=1#L190)

Unfortunately, `cargo-embed` (or rather `gdb-server`) dropped support for this command in 0.13.0 (https://github.com/probe-rs/probe-rs/issues/1564). Given that book [recommends installing `cargo-embed >= 0.11.0`](https://github.com/rust-embedded/discovery/blob/7d879f301dc364b3858eff6fbb90514c50587e91/microbit/src/03-setup/README.md?plain=1#L33), readers might benefit from a warning about this.

I've also had issues installing `cargo-embed` 0.11.0 or 0.12.0 due to dependencies being yanked, but this might be a user error :)

arr-ee, 2023-03-12:
I‘m happy to open a PR warning about this + hopefully https://github.com/probe-rs/probe-rs/pull/1565 lands, and we can just point people to the latest version.

PandaLord, 2023-04-20:
@arr-ee Thanks for your help and this coufused me a lot. I think this may need to be mentioned in the content.

commit 9c6ddba8 (2023-04-23): Bump `cargo-embed` version in setup to 0.18.0

Closes #515
commit d2ac71b7 (2023-04-23): Merge #521

521: Bump `cargo-embed` version in setup to 0.18.0 r=eldruin a=arr-ee

Closes #515

I decided to not include a note about checking `cargo-embed` version in case `monitor reset` does not work, version range should be enough imo.
PR #521 (2023-04-23): Bump `cargo-embed` version in setup to 0.18.0
Closes #515

I decided to not include a note about checking `cargo-embed` version in case `monitor reset` does not work, version range should be enough imo.
closed 2023-04-23 by bors[bot]