rust-embedded/discovery #517: Chapter 5.3: unexpected `SIGINT`s in gdb on M1 Mac

arr-ee, 2023-03-12:
I am not sure if this causes any actual issues as of right now, so this might be mostly to get pointers for further digging.

When going through the `Debug it` section, I'm seeing 

a) unexpected `SIGINT` right after `break main`
b) `SIGINT` when reaching line 16 (== `loop {}`)

I also noticed that if I start `stepi` after breaking on `break main`, I get `SIGINT` after every command.

Full GDB session:

```
❯ arm-none-eabi-gdb target/thumbv7em-none-eabihf/debug/led-roulette
GNU gdb (Arm GNU Toolchain 12.2 (Build arm-12.24)) 12.1.90.20221210-git
...
This GDB was configured as "--host=aarch64-apple-darwin20.6.0 --target=arm-none-eabi".
Reading symbols from target/thumbv7em-none-eabihf/debug/led-roulette...
(gdb) target remote :1337
Remote debugging using :1337
0x00000100 in nrf52833_pac::{impl#280}::fmt (self=0x281a7e10, f=0x9f867a82) at src/lib.rs:163
163	#[derive(Copy, Clone, Debug, PartialEq, Eq)]
(gdb) b main
Breakpoint 1 at 0x15c: file src/05-led-roulette/src/main.rs, line 9.
Note: automatically using hardware breakpoints for read-only addresses.
(gdb) c
Continuing.

Breakpoint 1, led_roulette::__cortex_m_rt_main_trampoline () at src/05-led-roulette/src/main.rs:9
9	#[entry]
(gdb) info thread
  Id   Target Id         Frame
* 1    Thread 1.1 (main) led_roulette::__cortex_m_rt_main_trampoline () at src/05-led-roulette/src/main.rs:9
(gdb) b 13
Breakpoint 2 at 0x168: file src/05-led-roulette/src/main.rs, line 13.
(gdb) c
Continuing.

Program received signal SIGINT, Interrupt.
led_roulette::__cortex_m_rt_main () at src/05-led-roulette/src/main.rs:10
10	fn main() -> ! {
(gdb) c
Continuing.

Breakpoint 2, led_roulette::__cortex_m_rt_main () at src/05-led-roulette/src/main.rs:13
13	    _y = x;
(gdb) c
Continuing.

Program received signal SIGINT, Interrupt.
led_roulette::__cortex_m_rt_main () at src/05-led-roulette/src/main.rs:16
16	    loop {}
```

Environment:

- `❯ sw_vers -productVersion`: `13.2.1`
- `❯ rustc -V`: `rustc 1.68.0 (2c8cc3432 2023-03-06)`
- `❯ cargo embed --version`: `cargo embed 0.16.0` (also tried with https://github.com/probe-rs/probe-rs/pull/1565 based off current `master`)
- `❯ arm-none-eabi-gdb --version`: `GNU gdb (Arm GNU Toolchain 12.2 (Build arm-12.24)) 12.1.90.20221210-git`
- microbit v2.21
- flashing command: `cargo embed --target thumbv7em-none-eabihf --features v2`

arr-ee, 2023-03-12:
I’m happy to dig further, but so far I was unable to understand where that message is even coming from — most of my experience with GDB came from couple of hours of reading while debugging this and #515 :)

arr-ee, 2023-03-12:
https://github.com/probe-rs/probe-rs/issues/1235 points out that microbit v2.20/v2.21 use different interface chip, and it _seems_ like it might behave slightly differently from v2.0. I have not looked into validating this hunch but leaving the link here just in case.

CarlosLaraFP, 2023-08-12:
Running into this same issue.

rodneylab, 2023-10-07:
I had this issue too.  The folks in the Rust Embedded Matrix room took a look, and could not see why it happens. However, you can just type `continue` in GDB, when you get the SIGINT, and carry on with the chapter.  They suggested using defmt with probe-rs for debugging embedded Rust code.