rust-embedded/discovery #531: [5.3] Different available breakpoints (dev build) compared to book

PieterJanVdb, 2023-06-20:
Not entirely sure what is happening here, I'm guessing it's some compiler optimization, but I can't seem to put a break on line 13 as shown in the chapter. Instead it only allows me to break on line 12 and line 16 (the loop).

The asm TUI looks like this:
[image]

While the disassembly looks like this:
[image]

At first glance it seems like the statement on line 13 has been optimized away, but I'm pretty sure the binary was built without optimizations:

[image]

For the record, I'm running this on Mac M1, with arm-none-eabi-gdb as the debugger

CarlosLaraFP, 2023-08-12:
Having this same issue on iMac.

kazauwa, 2023-09-08:
Same issue here on M1 Mac. The lines in the range 13-15 seem to be absent, you can verify this by executing `next` several times. You will see that gdb jumps from line 12 straight to line 16. Also, I tried disabling optimizations explicitly by specifying `"-C", "opt-level=0"` in `.cargo/config`, but it didn't help.

kazauwa, 2023-09-09:
I got it to work by downgrading rust to 1.66 and `half` crate to 2.2.1 (since 2.3.1 requires rust 1.70+). It seems that there was a change in 1.69 that is causing this weird optimization because 1.68.2 was the last version where I couldn't reproduce it. I wonder if this is the case for other platforms.

fosskers, 2023-09-30:
This is also currently happening on Arch Linux for me. For now I'll move forward with the chapter and "take his word for it".

fosskers, 2023-09-30:
Immediate discoveries: the lines in question aren't actually being compiled away. By manually `next`ing, while it does jump over the `_y = x`, doing a `info locals` afterwards does show that `_y = 42`, so the setting did indeed happen.

Berty48237, 2023-10-07:
I have the same issue on Linux Mint 20.04, Ubuntu 20.04. I.e. jumping from line 12 to 15 whilst single stepping.
I also have another problem with the code on section 5.4 'The led and delay abstractions'. When using 'next' to step over the call to let (mut delay, mut leds): (Delay, LedArray) = aux5::init() it jumps straight to the loop, as if I'd typed 'continue'.
When using the assembler view and 'si' to step into the aux5::init() function, it again steps over it.
Anyone else experiencing this?