rust-embedded/discovery #538: Requests for other language libraries

Jzow, 2023-07-31:
Can cc to the maintainer of other language support libraries after merging new chapters next time

For example, merging #535 and modifying the link, can we cc other language library maintainers to ensure that our translation version is synchronized.

Jzow, 2023-07-31:
@tomoyuki-nakabayashi @allynaell

eldruin, 2023-07-31:
Thank you for noticing this @Jzow.
This may be something that may be improved if we merge https://github.com/rust-embedded/book/pull/356 and if it goes well do the same for this book.
Do you have an opinion on that?

Jzow, 2023-08-01:
I agree that with the changes made to other embedded crates, it is inevitable for us to update the books

Jzow, 2023-08-02:
> Hi, @Jzow. Could you give me more context about this?

We are currently discussing how to synchronize our other translation work while updating official books without having to wait too long.

Regarding this issue:

The current idea of the resource community is to adopt  https://github.com/rust-embedded/book/issues/326 @mgeisle opinion and adopt a translation tool, but this requires the developer of the resource community to review it. Before this tool is officially used, we have to choose to submit a **pr** or **issue** to our translation library using **GitHub CI Tool** when other contributors perform pr on official books to ensure that we receive the information as soon as possible for modification