rust-embedded/discovery #539: Chapter 7.1: Error failed attaching to target

finleyfoxcraft, 2023-08-04:
In chapter 7.1, when I run `cargo embed --features v2 --target thumbv7em-none-eabihf`  I get the below error. 

```
~/discovery/microbit/src/07-uart/src$ cargo embed --features v2 --target thumbv7em-none-eabihf
    Finished dev [unoptimized + debuginfo] target(s) in 0.02s
      Config default
      Target /home/user/discovery/microbit/target/thumbv7em-none-eabihf/debug/uart
        WARN probe_rs::architecture::arm::memory::romtable > Component at 0xe0001000: CIDR0 has invalid preamble (expected 0xd, got 0x0)
        WARN probe_rs::architecture::arm::memory::romtable > Component at 0xe0001000: CIDR2 has invalid preamble (expected 0x5, got 0x0)
        WARN probe_rs::architecture::arm::memory::romtable > Component at 0xe0001000: CIDR3 has invalid preamble (expected 0xb1, got 0x0)
        WARN probe_rs::architecture::arm::memory::romtable > Component at 0xe0000000: CIDR0 has invalid preamble (expected 0xd, got 0xb1)
        WARN probe_rs::architecture::arm::memory::romtable > Component at 0xe0000000: CIDR1 has invalid preamble (expected 0x0, got 0x1)
        WARN probe_rs::architecture::arm::memory::romtable > Component at 0xe0000000: CIDR2 has invalid preamble (expected 0x5, got 0xb1)
        WARN probe_rs::architecture::arm::memory::romtable > Component at 0xe0040000: CIDR0 has invalid preamble (expected 0xd, got 0x0)
        WARN probe_rs::architecture::arm::memory::romtable > Component at 0xe0040000: CIDR2 has invalid preamble (expected 0x5, got 0x0)
        WARN probe_rs::architecture::arm::memory::romtable > Component at 0xe0040000: CIDR3 has invalid preamble (expected 0xb1, got 0x0)
       Error failed attaching to target
             
             Caused by:
                 0: Unable to load specification for chip
                 1: The connected chip could not automatically be determined.
```

I've followed the other chapters without issue, and double-checked that my Embed.toml is correct. I'm using the microbit v2.

adamgreig, 2023-08-04:
In your `Embed.toml`, have you uncommented the relevant `chip = "nrf52833_xxAA"` line? From the error message it looks like no chip has been selected and it failed to auto-detect one.

https://github.com/rust-embedded/discovery/blob/master/microbit/src/03-setup/Embed.toml#L5

finleyfoxcraft, 2023-08-04:
Yes, I've uncommented that line.

finleyfoxcraft, 2023-08-04:
Ah, wait.
I was running `cargo embed` from the `src` directory instead of the parent directory :roll_eyes: 
Apologies for taking your time.

closed 2023-08-04 by finleyfoxcraft