rust-embedded/discovery #540: This guide probably only works for Linux

tieje, 2023-08-20:
I tried using an M1 Mac and a Windows laptop to complete this tutorial.
Neither worked.
MacOS uses LLDB instead of GDB. LLDB breaks immediately when connecting to the GDB server.
Windows can't even connect to the GDB server using: `arm-none-eabi-gdb ../../target/thumbv7em-none-eabihf/debug/led-roulette`

I have not tried on Linux yet but it probably only works there.

adamgreig, 2023-08-20:
It should work fine on WIndows and MacOS. On MacOS you'll need to use gdb rather than lldb, but you can install arm-none-eabi-gdb or gdb-multiarch on MacOS. On Windows what error are you getting?

Have you followed the "Setting up a development environment" part of the book? [here](https://docs.rust-embedded.org/discovery/f3discovery/03-setup/index.html) for the STM32 version or [here](https://docs.rust-embedded.org/discovery/microbit/03-setup/index.html) for the nRF version. It has different instructions for Linux, Windows, and MacOS.

If there is some error in the description of how to get set up on one of those operating systems, that's a bug that we'll need to fix - it should work on all three.

neerajadsul, 2023-08-24:
On macOS with Apple Silicon (M1, M2 etc.), one needs to install architecture specific `gdb`. 
Therefore, you will need to install ` arm-none-eabi-gdb` speifically.

Details are at its formula: 
``` https://formulae.brew.sh/formula/arm-none-eabi-gdb#default```
For quick ref:
```brew install arm-none-eabi-gdb```

tieje, 2023-08-25:
Thank you, this worked. `layout asm` looks different but that's probably due to the architecture.
Should I create I a PR or do you want to handle it?

05-led-roulette % arm-none-eabi-gdb target/thumbv7em-none-eabihf/debug/led-roulette
(gdb) target remote :1337
Remote debugging using :1337
0x00000100 in nrf52833_pac::{impl#280}::fmt (self=0xb7f40fbc, f=0x24d0ee32)
    at src/lib.rs:163
163	#[derive(Copy, Clone, Debug, PartialEq, Eq)]

neerajadsul, 2023-08-25:
Glad it helped. Sure I will update and raise a PR. Thanks.

eldruin, 2023-08-28:
Should be fixed after #541. Thank you everybody!

closed 2023-08-28 by eldruin