rust-embedded/discovery #551: 3,4 Can't Verify Install - Failed Attaching To Target

datsimps, 2024-01-03:
I have tried verifying the install a couple dozen times, in hopes it was some witch craft preventing me from attaching to target. 
I am using WSL which you need to bind the usb device to the WSL, check
Make sure the usb has the correct permissions inside the WSL, check.
Make sure the probes can be found. check.
I come across this error to no avail:
[image]
The toml is the correct chip version but I have a sneaking suspicion something about the file isn't correct. 
[image]

I would appreciate any and all help on the matter. Thank you.

ajundi, 2025-05-25:
TLDR SOLUTION : workaround  bug assign RUSTFLAGS manually instead of config.toml  
`RUSTFLAGS="-C link-arg=-Tlink.x" cargo embed`

Hi I had the same issue on linux and I found out the problem after 2 days of debugging.  I came to report it but I saw you open issue so I will do that here. 

This is not a problem connecting to the USB device. The issue boils down to the elf file not being correct and probe-rs not finding anything to load onto the microcontroller. This is a BUG the actual linker is not using rustflags defined in the config.toml file in LINUX only (it is working fine in windows with same code and rustc version). 
I ran the same project in linux and windows and displayed the linker arguments using `cargo rustc -- --print=link-args` my `config.toml` inside .cargo file contains the following.
```
[target.'cfg(all(target_arch = "arm", target_os = "none"))']
rustflags = ["-C", "link-arg=-Tlink.x"]

[build]
target = "thumbv7em-none-eabihf"
```
In windows I got the following linker args at the end of the linker command
`... target\\thumbv7em-none-eabihf\\debug\\deps\\i2c-4754388f49bb4437" "--gc-sections" "-Tlink.x"`

In linux this is what I get at the end of my linker args

`... target/thumbv7em-none-eabihf/debug/deps/i2c-ce1445d509cca9ef" "--gc-sections"`

I checked the output in linux 

`readelf -l  ../../target/thumbv7em-none-eabihf/debug/i2c`

```
Elf file type is EXEC (Executable file)
Entry point 0x0
There are 3 program headers, starting at offset 52

Program Headers:
  Type           Offset   VirtAddr   PhysAddr   FileSiz MemSiz  Flg Align
  PHDR           0x000034 0x00010034 0x00010034 0x00060 0x00060 R   0x4
  LOAD           0x000000 0x00010000 0x00010000 0x00094 0x00094 R   0x10000
  GNU_STACK      0x000000 0x00000000 0x00000000 0x00000 0x00000 RW  0

 Section to Segment mapping:
  Segment Sections...
   00     
   01     
   02     
```
and In windows

`readelf.exe -l ..\..\target\thumbv7em-none-eabihf\debug\i2c    `
```
Elf file type is EXEC (Executable file)
Entry point 0x101
There are 5 program headers, starting at offset 52

Program Headers:
  Type           Offset   VirtAddr   PhysAddr   FileSiz MemSiz  Flg Align
  LOAD           0x010000 0x00000000 0x00000000 0x00100 0x00100 R   0x10000
  LOAD           0x010100 0x00000100 0x00000100 0x0aedc 0x0aedc R E 0x10000
  LOAD           0x01afdc 0x0000afdc 0x0000afdc 0x03054 0x03054 R   0x10000
  LOAD           0x020000 0x20000000 0x20000000 0x00000 0x00444 RW  0x10000
  GNU_STACK      0x000000 0x00000000 0x00000000 0x00000 0x00000 RW  0

 Section to Segment mapping:
  Segment Sections...
   00     .vector_table
   01     .text
   02     .rodata
   03     .bss
   04
```

WORKAROUND: 
Manually assign RUSTFLAGS before running the build command 
 `RUSTFLAGS="-C link-arg=-Tlink.x" cargo rustc -- --print=link-args`

This displayed the correct linker arguments in linux and probe-rs worked great.
`......target/thumbv7em-none-eabihf/debug/deps/i2c-9061183c020b57e6" "--gc-sections" "-Tlink.x"`

DanielSimpso, 2025-05-25:
I appreciate the information! I wish there was a method to pin this to the README.md to help everyone out. Very informative and thorough, keep posting your findings for everyone else man!

datsimps, 2025-05-25:
Much appreciated information! I wish I could post this to the README.md, for everyone else. Very informative and thorough, keep spreading the word lol