rust-embedded/discovery #564: [Chapter 5.3] Continue hangs, stuck at "0x00000000 in __vector_table ()"

garrelj1, 2024-04-15:
I'm able to build and flash the code. Once I try to debug, I get the following:

```
(gdb) target remote :1337
Remote debugging using :1337
warning: while parsing target description (at line 4): Target description specified unknown architecture "armv7"
warning: Could not load XML target description; ignoring
0x00000000 in __vector_table ()
```

Notice `0x00000000 in __vector_table ()`

After adding a break point, and continuing, it just hangs:

```
(gdb) b main
Breakpoint 1 at 0x15c: file src/05-led-roulette/src/main.rs, line 9.
Note: automatically using hardware breakpoints for read-only addresses.

(gdb) continue
Continuing.

```

I've also tried `step`, which gives

```
(gdb) step
Single stepping until exit from function __vector_table,
which has no line number information.

Program received signal SIGINT, Interrupt.
0x00000000 in __vector_table ()
```

And I've tried `stepi`, which gives

```
(gdb) stepi

Program received signal SIGINT, Interrupt.
0x00000000 in __vector_table ()
```

Any ideas of how else to debug or get further?

Michal2404, 2024-09-27:
Hey, Did you manage to resolve this issue? I am currently hanging on it and have no idea what is the cause...

BartMassey, 2024-09-27:
Could you please take a look at https://docs.rust-embedded.org/discovery-mb2 and https://github.com/rust-embedded/discovery-mb2 and see if you can reproduce this issue there? If so, please file an issue there and we'll try to get it figured out and fixed.

Michal2404, 2024-09-27:
Have you tried running **gdb-multiarch** instead of **gdb**? I had the same issue and it helped me.

zanebliss, 2024-10-10:
@Michal2404 What did you do exactly to fix this? Did you use `gbd-multiarch` as the command to run instead of `gdb`?

Michal2404, 2024-10-10:
Exactly. It depends on which version of gdb you have installed (chapter 3 of the book).

garrelj1, 2025-09-26:
I haven't tried myself (been working other stuff), but others seem to have gotten this working using `gdb-multiarch`. I'll close the issue

closed 2025-09-26 by garrelj1