rust-embedded/discovery #568: Extract and archive support for microbit:v1

hdoordt, 2024-05-13:
Currently, the new book supports both the Micro:bit V1 and the V2. Both are based on a different MCU and so the book contains duplicate examples supporting both. As the V1 is basically not sold much anymore, supporting it seems to add more maintanance burden and, more importantly, noise to the reader than necessary. 

I'd like for the content Micro:bit V1 to be extracted to a separate book in a separate repo, and archive that. The goal is for the book in this repo to only support Microbit:V1

sirhcel, 2024-05-14:
As this discussion started here, I'd like to bring over my thoughts for today's meeting.

What do you see as the benefit from having a separate repo? If it is about just archiving the contents for the micro:bit V1 and the STM32F3 (#567 which i would like to address here as well) what about keeping it in this repo under a certain tag or branch? We could mention this briefly in a note on previous versions of the book.

To me a more prominent question would be whether to keep the rendered version of the current book online. This looks somewhat charming to me:

* Existing links would not break
* This would not confront people on this journey having already some cognitive load with hitting a HTTP 404 "wall"
* We could add a note on each page for guiding them gently to the new version of the book

Have we decided about how to maintain the current version of the book yet? A complete freeze still fixing grave things? The latter would not include keeping the documentation up to the latest versions of tooling.

How to deal with the currently open issues and PRs? What about using note or issue template for pointing out that the current version is about to be deprecated soon and stating the policy on maintaining the old version to set some expectations for contributors? What about closing current issues and PRs once the upcoming update made some significant progress?

hdoordt, 2024-05-22:
Discussion has moved to https://github.com/rust-embedded/wg/pull/759

closed 2024-05-22 by hdoordt