rust-embedded/discovery #579: probe-rs-tools installation fails

dyozie, 2024-08-12:
In Ubuntu, installing probe-rs-tools with cargo as indicated at https://github.com/rust-embedded/discovery/blob/71004431ef2ec0663f137a7a6262f8e5e1be458d/microbit/src/03-setup/README.md?plain=1#L72 fails with an error:

error[E0252]: the name `FromBytes` is defined multiple times

According to the latest probe-rs release notes, the toolkit should instead be installed via a shell/powershell script.  https://github.com/probe-rs/probe-rs/releases has the relevant commands.

BartMassey, 2024-08-13:
Thanks much for bringing this to our attention! That said, I'm not sure how to proceed.

It looks like the method for installing the `probe-rs` tools has again changed: we are now expected to download release binary tarballs from the `probe-rs` host and install them with a script. This seems to me not OK to recommend, at least without some clear explanation. The release notes refer me to the homepage or README, but I can't find any information there about this.

Current `probe-rs-tools` from top-of-tree `master` on their Git repo builds and installs fine. It looks like commit 6e975baf fixed the issue, but no release to `crates.io` has been made since then.

Several choices here: guidance appreciated:
* Make the instructions just refer the user to <https://probe.rs> for installation instructions. This is probably the safest choice, as I imagine there will be more `probe-rs` installation issues in the future.
* Make the instructions pull a known-good commit of `probe-rs-tools`:

         cargo install --git=https://github.com/probe-rs/probe-rs --rev=aa4e7c09 probe-rs-tools
         
  This would get us a "known-quantity" version of `probe-rs`, but would miss out on future enhancements
* Get the `probe-rs` folks to release 0.24.1 with the bug fixed. We should maybe then do

          cargo install --version='^0.24' probe-rs-tools

   so that subsequent minor versions will work, or even give up and

          cargo install probe-rs-tools
          
   and hope.

* Something else?

BartMassey, 2024-08-13:
Threw a bunch of stuff in the air. We'll see how it falls.

BartMassey, 2024-08-13:
Turns out that `cargo install --locked probe-rs-tools` resolves the issue. Also turns out that the `probe-rs` folks don't want people to install using `cargo install`, for some reason I haven't yet determined.

dyozie, 2024-08-13:
Thanks @BartMassey  - just confirming that installing the locked version works for me on another system. That seems like the quickest/simplest fix for the tutorial. I'm not sure what the tradeoff would be for using the locked build vs the official installer.

BartMassey, 2024-08-13:
The larger problem is that probe-rs apparently released 0.24.0 to crates.io "by mistake" and doesn't plan to release future versions in this way. I too don't understand what tradeoffs prompted this.

That said, I'm glad this worked for you and have a PR to adjust the Book accordingly. Thanks much.

spyoungtech, 2024-08-30:
While this issue is open... it is also worth mentioning the guide also neglects to include installing `cmake` and Visual Studio, which is apparently required to compile `probe-rs`?

Initially got an error saying "program not found. Is cmake installed?" then after installing cmake with `winget`, I got the following error:

```
  --- stderr
  CMake Error at CMakeLists.txt:35 (project):
    Generator

      Visual Studio 17 2022

    could not find any instance of Visual Studio.

  thread 'main' panicked at C:\Users\Spencer\.cargo\registry\src\index.crates.io-6f17d22bba15001f\cmake-0.1.50\src\lib.rs:1098:5:
```

Although I have visual studio enterprise 2022 installed with a rather full complement of components, I get this problem. So, not sure what to do about this. As it stands, I cannot move on with the tutorial.

spyoungtech, 2024-08-30:
Circling back around to this, I was able to resolve this by upgrading rust from 1.76.0 to 1.80.1 -- then the `--locked` install worked.

The guide currently says Rust 1.57.0 or newer should work, but it seems like an even newer version of rust is actually needed to get probe-rs installed now.

jannic, 2024-08-30:
It looks like `offset_of!` was stabilized in 1.77.0: https://releases.rs/docs/1.77.0/
(Usage of that macro was added in probe-rs v0.24.0)

msilveus, 2025-07-06:
[image] 

I got this error when I ran cargo install probe-rs-tools

BartMassey, 2025-07-07:
> I got this error when I ran cargo install probe-rs-tools

Sigh. This is almost certainly a false positive from Norton AV.

There is very little we can do about this. You can try filing an issue against `probe-rs`.

If you clone the repo do you get the same error when running `cargo install -path .`? This would make it more likely that the `probe-rs` folks would be interested.

For now, I suggest installing binaries as described on the `probe-rs` website and seeing if Norton will take that.