rust-embedded/discovery #586: docs, Setting Up page, explain why the cargo installs are pinned?

sprive, 2024-11-25:
Following [Setting Up page](https://docs.rust-embedded.org/discovery/microbit/03-setup/index.html), the tools versions are specified in _pinned fashion.._
Example: 

```
...
$ cargo install cargo-binutils --vers 0.3.3

$ cargo size --version
cargo-size 0.3.3
```

Are these pins required, and if so: for what reason? the latest version is .3.6 for example.

BartMassey, 2024-11-25:
Those are the versions we successfully used when this was written. Check out https://github.com/rust-embedded/discovery-mb2 for the more recent version of the book, which has more recent versions of the dependencies. You're right that some text would be good there.

sprive, 2024-12-06:
Awesome, thanks. I am composing a docs PR to capture your explanation. BIAB

sprive, 2024-12-06:
Awesome.   That pointer helps (I had the newer v2 micro:bit, but I stumbled into the older guide).  TY.