rust-embedded/discovery #587: Chapter 5 LED roulette light it up failed unwinding panics are not supported without std

acyanbird, 2024-11-26:
Hi I've stucked with the chapter 5, when I replace the code on book and put it in main, executing cargo embed but failed.

```
 cargo embed
   Compiling led-roulette v0.1.0 (/home/acy/rust/discovery/microbit/src/05-led-roulette)
error: could not compile `led-roulette` (bin "led-roulette") due to 2 previous errors
error: unwinding panics are not supported without std
  |
  = help: using nightly cargo, use -Zbuild-std with panic="abort" to avoid unwinding
  = note: since the core library is usually precompiled with panic="unwind", rebuilding your crate with panic="abort" may not be enough to fix the problem

error: aborting due to 1 previous error

       Error Failed to run cargo build: exit code = Some(101).
```
I have the rust in stable channel with version
```
 rustc -V
rustc 1.82.0 (f6e511eec 2024-10-15)
```
I try to add 
```
[profile.dev]
panic = "abort"
```
in cargo.toml but still not working, could anyone tell me how to continue?

acyanbird, 2024-11-27:
After adding this to cargo.toml at root
```
[profile.dev]
panic = 'abort'
```
It works a bit, but another error appear
```
 cargo embed
   Compiling led-roulette v0.1.0 (/home/acy/rust/discovery/microbit/src/05-led-roulette)
error: could not compile `led-roulette` (bin "led-roulette") due to 2 previous errors
error: linking with `cc` failed: exit status: 1
  |
  = note: LC_ALL="C" PATH="/home/acy/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/bin:/home/acy/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/bin:/home/acy/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/bin:/home/acy/.bun/bin:/home/acy/qemu-7/qemu7-path:/home/acy/.cargo/bin:/home/acy/.bun/bin:/home/acy/qemu-7/qemu7-path:/home/acy/.cargo/bin:/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games:/home/lucia/.local/bin:/home/acy/.local/share/JetBrains/Toolbox/scripts:/home/lucia/.local/bin:/home/lucia/2023cse/compiler/llvmb/bin:/home/lucia/.local/bin:/home/lucia/2023cse/compiler/llvmb/bin" VSLANG="1033" "cc" "-m64" "/tmp/rustcVmxSsR/symbols.o" "/home/acy/rust/discovery/microbit/target/debug/deps/led_roulette-f7eeb233d1f965a3.10z0zbj18yqo56cyip92t02pm.rcgu.o" "/home/acy/rust/discovery/microbit/target/debug/deps/led_roulette-f7eeb233d1f965a3.5bqykmjxflxueshdz83rc8afn.rcgu.o" "-Wl,--as-needed" "-Wl,-Bstatic" "/home/acy/rust/discovery/microbit/target/debug/deps/libmicrobit-d2170b61ea424045.rlib" "/home/acy/rust/discovery/microbit/target/debug/deps/libmicrobit_common-ef0009c276df37ff.rlib" "/home/acy/rust/discovery/microbit/target/debug/deps/libtiny_led_matrix-acf4bc74c3a9c058.rlib" "/home/acy/rust/discovery/microbit/target/debug/deps/libnrf52833_hal-4e0b71c843aaaaca.rlib" "/home/acy/rust/discovery/microbit/target/debug/deps/libnrf_hal_common-b5ad5722fe151f21.rlib" "/home/acy/rust/discovery/microbit/target/debug/deps/libnrf_usbd-44250aee3cf8a0b7.rlib" "/home/acy/rust/discovery/microbit/target/debug/deps/libusb_device-50f3586dfb9e09bd.rlib" "/home/acy/rust/discovery/microbit/target/debug/deps/libcritical_section-fe74c5fe622d3e62.rlib" "/home/acy/rust/discovery/microbit/target/debug/deps/libcritical_section-292ae6aed9aa1699.rlib" "/home/acy/rust/discovery/microbit/target/debug/deps/libbare_metal-702897061ca01a79.rlib" "/home/acy/rust/discovery/microbit/target/debug/deps/libfixed-1a4d6c23ff5bb423.rlib" "/home/acy/rust/discovery/microbit/target/debug/deps/libtypenum-bf88f37685656557.rlib" "/home/acy/rust/discovery/microbit/target/debug/deps/libbytemuck-57888f3f375647db.rlib" "/home/acy/rust/discovery/microbit/target/debug/deps/libhalf-a839ec135333e37b.rlib" "/home/acy/rust/discovery/microbit/target/debug/deps/libaz-46f5cf315c5c5956.rlib" "/home/acy/rust/discovery/microbit/target/debug/deps/librand_core-5607b160bd39df79.rlib" "/home/acy/rust/discovery/microbit/target/debug/deps/libcfg_if-356635494db797cd.rlib" "/home/acy/rust/discovery/microbit/target/debug/deps/libembedded_storage-8345e4bfd48f7f07.rlib" "/home/acy/rust/discovery/microbit/target/debug/deps/libembedded_dma-f8ce578040dae943.rlib" "/home/acy/rust/discovery/microbit/target/debug/deps/libstable_deref_trait-54d3748517d1c555.rlib" "/home/acy/rust/discovery/microbit/target/debug/deps/libcast-7ee0d05f2944fa85.rlib" "/home/acy/rust/discovery/microbit/target/debug/deps/libnrf52833_pac-79908c2fab7d63cf.rlib" "/home/acy/rust/discovery/microbit/target/debug/deps/libcortex_m-5396d976247ea9f2.rlib" "/home/acy/rust/discovery/microbit/target/debug/deps/libvolatile_register-9e5c74df487eea8e.rlib" "/home/acy/rust/discovery/microbit/target/debug/deps/libvcell-09d0d6ae2c984ac8.rlib" "/home/acy/rust/discovery/microbit/target/debug/deps/libbare_metal-b9a665e92edf52c8.rlib" "/home/acy/rust/discovery/microbit/target/debug/deps/libembedded_hal-dfcdb4a76a391c76.rlib" "/home/acy/rust/discovery/microbit/target/debug/deps/libvoid-635b1b893e73aa30.rlib" "/home/acy/rust/discovery/microbit/target/debug/deps/libnb-ea348ba05adc047c.rlib" "/home/acy/rust/discovery/microbit/target/debug/deps/libnb-71efebda3fcc70b1.rlib" "/home/acy/rust/discovery/microbit/target/debug/deps/libpanic_halt-5f09d18042b08d0e.rlib" "/home/acy/rust/discovery/microbit/target/debug/deps/libcortex_m_rt-37f945618b72b9e7.rlib" "/home/acy/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/lib/librustc_std_workspace_core-7874c355ab5ed077.rlib" "/home/acy/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/lib/libcore-46b25da62cc69443.rlib" "/home/acy/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/lib/libcompiler_builtins-c71612932829263c.rlib" "-Wl,-Bdynamic" "-Wl,--eh-frame-hdr" "-Wl,-z,noexecstack" "-L" "/home/acy/rust/discovery/microbit/target/debug/build/led-roulette-ec32d443190233c2/out" "-L" "/home/acy/rust/discovery/microbit/target/debug/build/cortex-m-rt-1826f760730b43f8/out" "-L" "/home/acy/rust/discovery/microbit/target/debug/build/nrf52833-hal-f40de9d4f3d05c58/out" "-L" "/home/acy/rust/discovery/microbit/target/debug/build/nrf52833-pac-a24b50b11d7ec32b/out" "-L" "/home/acy/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/lib" "-o" "/home/acy/rust/discovery/microbit/target/debug/deps/led_roulette-f7eeb233d1f965a3" "-Wl,--gc-sections" "-pie" "-Wl,-z,relro,-z,now" "-nodefaultlibs"
[... cut ...]
  = note: some `extern` functions couldn't be found; some native libraries may need to be installed or have their path specified
  = note: use the `-l` flag to specify native libraries to link
  = note: use the `cargo:rustc-link-lib` directive to specify the native libraries to link with Cargo (see https://doc.rust-lang.org/cargo/reference/build-scripts.html#rustc-link-lib)

error: aborting due to 1 previous error

       Error Failed to run cargo build: exit code = Some(101).
```

It seems that some lib cannot be found?

BartMassey, 2024-11-27:
Sorry things are going rough! Did you work through the previous chapters?

I think you're not getting the target architecture set up right somehow. Or something.

Please try again with https://docs.rust-embedded.org/discovery-mb2/ and drop an issue on that repo if you still are having trouble.

acyanbird, 2024-11-27:
Oh it works with this version! I guess is the .cargo/config.toml in this repo with flag
```
[target.thumbv7em-none-eabihf]
runner = "probe-rs run --chip nRF52833_xxAA"
rustflags = [
  "-C", "linker=rust-lld",
  "-C", "link-arg=-Tlink.x",
]
```
solve the problem. Thank you so much!!!

closed 2024-11-27 by acyanbird